hathach/tinyusb #1497: Device Address or EP index greater than 15 will cause OHCI driver to crash

Ryzee119, 2022-06-07:
### Operating System

Windows 11

### Board

OHCI

### Firmware

This index is only 4 bits and stores the device address or an index of an endpoint in `ed_pool`. As soon as the value in index is >15 it will overflow and cause OHCI hardware to crash.

I can see we've had to squeeze it in the reserved area to fit within a `gtd`, so not sure on a good fix for it. Ive confirmed this is the issue by temporarily increasing the `index` field size, by reducing the `expected_bytes` field.

https://github.com/hathach/tinyusb/blob/896c707cd4e0a07fcfc105e793ede134398f290b/src/portable/ohci/ohci.h#L71

https://github.com/hathach/tinyusb/blob/896c707cd4e0a07fcfc105e793ede134398f290b/src/portable/ohci/ohci.c#L486

### What happened ?

Would prevent all future control transfers from completing after `Set Address = 16` request completed.

### How to reproduce ?

Plug in a number of devices until a usb hub or devices is allocated an address of 16 or more.

The easiest way is to set `CFG_TUH_DEVICE_MAX` to 15, then the first USB hub connected will be given an address of 16. It will fail on the first control transfer after the Set Address command is complete

### Debug Log as txt file

```
HID init
HUB init
[0:0:0] USBH DEVICE ATTACH
Full Speed
[0:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
dev 00 on EP 00 with 8 bytes
dev 00 on EP 80 with 8 bytes
[0:0] Control data:
  0000:  12 01 10 01 09 00 00 08                          |........|
dev 00 on EP 00 with 0 bytes

Set Address = 16
[0:0] Set Address: 00 05 10 00 00 00 00 00
dev 00 on EP 00 with 8 bytes
dev 00 on EP 80 with 0 bytes

[0:16] Open EP0 with Size = 8
Get Device Descriptor 16
[0:16] Get Descriptor: 80 06 00 01 00 00 12 00
dev 00 on EP 00 with 8 bytes
.... never completes
```

hathach, 2022-06-08:
you are right, I have written this long ago for NXP LPC MCU, and never thought it would need to support more than 16 devices. I actually prefer the solution 2, even though we need a few extra code to access the custom data array (probably 2 bytes per TD), but in general it save lots more SRAM. It may not be a lot but less is better since TinyUSB is expected to run on small MCU as well. 

> I propose a couple solutions:
>
> - Expand the td struct to include this custom data. To maintain 16byte aligned, we should have to expand the struct to 32 bytes so would waste a bit of space but provide plenty of room to store al lthe info we could ever need.
> - Replace the reserved area with an index that points to an index in a custom data array that can hold this info. We would have to manage another array which is a bit messier.

Ryzee119, 2022-06-08:
I will create a PR for this fix soon
Thanks for your input 👍

I am using 4 ports, 5 hubs and potentially 12 devices so pushing the limits a bit.