hathach/tinyusb #1520: SAMx7x CDC-MSC example does not link with LOG=2

tmplt, 2022-06-17:
### Operating System

Linux

### Board

SAM V71 Xplained Pro

### Firmware

`examples/device/cdc_msc` @ e0d7c16788f15c72c5a7837656d8b6d3ed28c3e0

### What happened ?

Upon building with
```shell
$ cd examples/device/cdc_msc
$ make BOARD=same70_xplained LOG=2 all
```
linking fails with
```
LINK _build/same70_xplained/cdc_msc.elf
/nix/store/l3rvj8w9ci8wj34kmsvga1ijikmngz0y-gcc-arm-embedded-10.3.1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /nix/store/l3rvj8w9ci8wj34kmsvga1ijikmngz0y-gcc-arm-embedded-10.3.1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
collect2: error: ld returned 1 exit status
make: *** [../../rules.mk:100: _build/same70_xplained/cdc_msc.elf] Error 1
```

Building without `LOG=2` links as expected, and the example works on the V71 even though the example is written for the E70.

### How to reproduce ?

```sh
$ git clone https://github.com/hathach/tinyusb.git && cd tinyusb
$ git submodule update --init
$ cd examples/device/cdc_msc
$ make BOARD=same70_xplained LOG=2 all
[-Wcast-qual warnings...]
LINK _build/same70_xplained/cdc_msc.elf
/nix/store/l3rvj8w9ci8wj34kmsvga1ijikmngz0y-gcc-arm-embedded-10.3.1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /nix/store/l3rvj8w9ci8wj34kmsvga1ijikmngz0y-gcc-arm-embedded-10.3.1/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
collect2: error: ld returned 1 exit status
make: *** [../../rules.mk:100: _build/same70_xplained/cdc_msc.elf] Error 1
```

tmplt, 2022-06-17:
Possibly a toolchain issue, but I do not have a non-NixOS system at hand to verify.

perlindgren, 2022-06-17:
I tested under arch (manjaro).

After a bit of tool-chain fiddling I was able to reproduce the error:
```
LINK _build/same70_xplained/cdc_msc.elf
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
collect2: error: ld returned 1 exit status
make: *** [../../rules.mk:100: _build/same70_xplained/cdc_msc.elf] Error 1
```

hathach, 2022-06-24:
this is usually missing symbol end/_end in the linker script. The end symbol is used by nanolib for printf().

hathach, 2022-06-29:
this is fixed by adding end symbol to linker of sam 7x in microchip driver submodule
https://github.com/hathach/microchip_driver/commit/9e8b37e307d8404033bb881623a113931e1edf27

and is resolved by #1539