hathach/tinyusb #1528: synopsys dcd_dwc2 device mode bug

ClarkyKent, 2022-06-22:
### Operating System

Linux

### Board

STM32H743x

### Firmware

I am using custom FW where FreeRTOS is used and have 2 CDC in device mode.  Vbus sensing is enabled.

### What happened ?

If I power on the board with USB already connected. Everything is fine. 2 USB com ports get detected on host machine(Linux/Windows) and everything woks fine as intended. 
But if I power on our board without USB(as board has a battery as main source of power) and I do not attach USB, STM32H743 resets itself as soon as it executes connect `dcd_connect(rhport);`

### How to reproduce ?

In order to reproduce this bug, it is important to have a board which is powered by a battery instead of USB. and then run any example from example. STM32h7 should reset when it tries to connect to USB after `dcd_connect(rhport);` command.

ClarkyKent, 2022-06-22:
I have kind of found the solution to this. so replying here. 

Once I have removed this following line in function `dcd_init()` everything workas as inteneded.
`// Clear A override, force B Valid`
`//dwc2->gotgctl = (dwc2->gotgctl & ~GOTGCTL_AVALOEN) | GOTGCTL_BVALOEN | GOTGCTL_BVALOVAL;`

Even-though MCU is powered on and stack BLE is initialized, it does not resets itself anymore. and if i connect an USB , it starts to enumerate and it shows USB device on HOST machine. 

I do not know if forcing B valid is necessary , and when i see the manual of STM32H7 it states that these registers are available only in Host mode. 

[image]

[image]

hathach, 2022-06-22:
> In order to reproduce this bug, it is important to have a board which is powered by a battery instead of USB. and then run any example from example. STM32h7 should reset when it tries to connect to USB after dcd_connect(rhport); command.

can you specify which example to run and on which specific dev board.

ClarkyKent, 2022-06-22:
Till now I tested on my custom PCB with STM32H743II. 

But I did a test on STM32H743ZI nucleo board with `cdc_msc_freertos` example. It is same , We can see LED(green) blinks(which means MCU is restarting/resets)

hathach, 2022-06-24:
also, on your custom board, did the reset happen with bare-metal example e.g cdc_msc (without freertos) ?