hathach/tinyusb #1547: Example for using multiple TMC ports on one device

diminDDL, 2022-07-07:
While trying to port a project from LUFA to tinyUSB I can't seem to find a way to open several TMC ports on a single device. For example the [cdc_dual_ports](https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_dual_ports) example had a function that allows the user to select whichever port would be addressed `uint32_t tud_cdc_n_write(uint8_t itf, void const* buffer, uint32_t bufsize)` I couldn't find anything similar for TMC however. I would like to be wrong but there does not appear to be any way to do what I am trying in tinyUSB right now. 

**Describe the solution you'd like**
The solution would be an example that shows how one can initiate and use several TMC ports on a single device.

pigrew, 2022-07-07:
Currently, the TMC code only supports a single interface. Its API would require modification to support multiple instances. This is low on my priority list (and there are other changes I'm planning to make first).

diminDDL, 2022-07-07:
Okay, thank you for the notice. I will wait then.

pigrew, 2022-07-07:
API-wise, I'm not sure which way to go.... The major question is if the application or TinyUSB should be responsible for the allocation of the class' data structures. I'm contemplating something for memory allocation like:

```
bool usbtmc_connect_cb(uint32_t config, uint32_t itf, struct tmc_state **state ); // app may malloc or return static memory address
void usbtmc_disconnect_cb(uint32_t config, uint32_t itf, struct tmc_state **state ); // app should free the memory here, if needed
```
where config and interface are the USB configuration and interface numbers being setup.

We could add a user_data element to `struct tmc_state`, which would be passed to the other TMC callbacks, instead of passing the config and interface to every call? I should do a better survey of other USB libraries to see how they are organized.

But, these don't really conform with how the rest of the library is setup at the moment. Thought? 

(Also, this work would be way out in the future, like many many months. )

pigrew, 2022-07-07:
It looks like that project only supports a single interface at a time.

I'm starting to believe that TMC is not a great protocol to use for USB-GPIB adapters. VXI-11.2 (or perhaps HiSLIP) is a better match. These could be done using ethernet over USB (RNDIS, ECM, etc...) Also note that each USB488 interface needs 2 endpoints, and most USB devices are limited to 6 or 8 endpoints... so 2 or 3 TMC interfaces per USB device.