hathach/tinyusb #1548: CDC doesn't respect no flow control setting (flow control always on)

MrGreensWorkshop, 2022-07-07:
### Operating System

Others

### Board

Raspberry pi pico

### Firmware

just simple app to test cdc behavior.

https://github.com/MrGreensWorkshop/RasPiPicoSDK_cdc_test

- Download
- Compile and run. (or just use binary named PicoTest.uf2)
- use Terminal.exe app to check DTR behavior.

### What happened ?

Handshake setting is set to none. But to receive data we need to enable to DTR. When handshake is off, we shouldn't be force to enable DTR to receive data. or any other flow control shouldn't be changed (ex DSR)

I believe this is the same problem with #872. maybe also related with #746

[image]

### How to reproduce ?

https://github.com/MrGreensWorkshop/RasPiPicoSDK_cdc_test

- Download
- Compile and run. (or just use binary named PicoTest.uf2)
- use Terminal.exe app to check DTR behavior as shown in gif file.

### Screenshots

This the demonstration that 2 FDTI USB serial converter connected each other using.
It shouldn't be forced to enable DTR to receive data either sides unless you enabled the hardware handshake.

```
Rx <- Tx
Tx -> Rx
GND <-> GND
```
[image]

TravisRo, 2022-07-10:
This has been done intentionally.  They use the DTR signal to detect when the PC/host has actually opened or closed the com port.  I agree, it causes problems it some cases.  Most of the arduino CDC implementations do this as well; it is fairly common.

I have a pull request submitted that adds a CH341 emulation class driver which would solve your problem.
[https://github.com/hathach/tinyusb/pull/1507](url)

You can get this from my fork here: [https://github.com/TravisRo/tinyusb/tree/master](url)

The api is nearly 100% compatible with the cdc driver except you have to change all of your "tud_cdc" prefixes to "tud_ch341"

MrGreensWorkshop, 2022-07-11:
Thank you for responding my inquiry. Yes I checked the code. And Thank you for the PR #1507

When people use MCU which support USB, and if serial port is needed using CDC is common to use instead of using USB serial converters. When it happens, you expect using CDC to be the same as using a USB serial converter. Using DTR to detect connection shouldn't be implemented. We can't expect all application to raise DTR when open serial port. That's why I raise this issue.

HiFiPhile, 2022-07-11:
@MrGreensWorkshop 
CDC class has very limited function. I made this issue long time ago https://github.com/hathach/tinyusb/issues/400 and had a long discussion https://github.com/hathach/tinyusb/pull/401

MrGreensWorkshop, 2022-07-11:
I saw in #482 too. I think author of this repo only tested in VT100 terminals like PuTTY or Teraterm. That's why he thinks its ok, but it's not.

PuTTY or Teraterm terminals are made for VT100 etc., not exactly plain text transmission. You can use if you want. But those are uses other pins too. DTR -> DSR and DSR <- DTR. That's the problem. 

Asking host application for DSR on is reduce the compatibility with other apps if user is explicitly select no flow control.

HiFiPhile, 2022-07-11:
I think you should ask pico SDK to remove tud_cdc_n_connected() test, just like #506 remove this test from examples.

MrGreensWorkshop, 2022-07-11:
pico SDK -> They don't even understand the problem. Check this [#906](https://github.com/raspberrypi/pico-sdk/issues/906)

MrGreensWorkshop, 2022-07-12:
I see your point.

MrGreensWorkshop, 2022-07-19:
@hathach 
I see `examples/device/cdc_msc` not using `tud_cdc_n_available`, but example `examples/device/cdc_dual_ports` uses `tud_cdc_n_available`

This is how raspberry pi pico sdk implemented. they are using `tud_cdc_n_available` that's why its happening.
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb.c

hathach, 2022-07-19:
> Kind of you are right it's their problem. I think there must be better implementation for tud_cdc_connected don't you think so?

Let me know if you figure out the better way to determine there is a connected client, I am happy to make the changes.

Note: there is a difference in cdc behavior when dtr is not set, the tx buffer will become overwritable, i.e if tx fifo is full, it will be overwritten. more details of this behavior is discussed in previous issue/pr mentioned above.

MrGreensWorkshop, 2022-07-19:
I assume you are referring #482 ?

MrGreensWorkshop, 2022-07-19:
I made some changes and test it's behavior.

https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb.c

first changed this
```
bool stdio_usb_connected(void) {
    return true;
}
```
and renamed `tud_cdc_connected()` to `stdio_usb_connected()`

When port is opened all data in tx buffer output to terminal.
This is not problem because when handling data on pc we can flush buffer.

Is there any other concerns?

hathach, 2022-07-19:
as mentioned above, the only difference in behavior is tx fifo can be overwritten when dtr is not set. When dtr is set, tinyusb will blocking wait for tx fifo to be sent since it is certain there is waiting client. In other words, If you send too many data in short period, there maybe some missing. 

connected() is option, going to support dtr or not is up to stdio integration i.e rpi team. For the changes you propose is within pico-sdk, you should ask them to see if it is acceptable or not.

MrGreensWorkshop, 2022-07-19:
@hathach Things are more clear now. Thank you for your explanation.

hathach, 2022-07-19:
no problem, I am glad that help. The current state of control signals (dtr,rts, cts etc...) in cdc acm are not standardized among terminal clients, though more and more terminal use dtr as connected nowsaday. I will try to revise/improve this later on when having time. I guess we could now close this issue here ?

MrGreensWorkshop, 2022-07-19:
I open the [this PR](https://github.com/raspberrypi/pico-sdk/pull/932) on pico-SDK.

Clearly `tud_cdc_connected` used by choice of pico-sdk team.

> though more and more terminal use dtr as connected nowadays.

About that. It is also mentioned in #482.

> Universal Serial Bus Communications Class Subclass Specification for PSTN Devices for Abstract Control Model (ACM) ..

PSTN Devices are mostly modems. Most of time we need flow control to control data flow. Let's not confuse with modems.

The reason that I raise the issue is not about the terminals default settings. It's about the application side. Most people use terminals for test purpose. 

As you know more mcu have usb support nowadays, so serial port replace with CDC. People like me want to use cdc without any change in host application. Host serial app could be run on PC, mobile device or host mcu.

Thank you again for this great project.

closed 2022-07-19 by MrGreensWorkshop