hathach/tinyusb #1558: tud_cdc_rx_cb is not called

DNedic, 2022-07-13:
### Operating System

Linux

### Board

STM32 boards

### Firmware

examples/device/cdc_msc_freertos

### What happened ?

Using the cdc_msc_freertos example compiled with either ESP-IDF v4.4.1 or latest master with an STM32 chip, when copying files to the MSC device the tud_cdc_rx_cb is not being called.

### How to reproduce ?

1. Grab the TinyUSB repo
2. Set up ESP-IDF and export environment
3. Build example using TinyUSB make build system
4. Start a debug session and place a breakpoint on tud_cdc_rx_cb
5. Copy a file over to the MSC device on a host PC

hathach, 2022-07-18:
Can you be specific on which 
- esp32 boards are testing with
- which example you are building

Howver, I don't see why would tud_cdc_rx_cb() has to be invoked when copying file to msc. cdc & msc are 2 separated classes.

DNedic, 2022-07-25:
> Can you be specific on which
> 
> * esp32 boards are testing with
> * which example you are building
> 
> Howver, I don't see why would tud_cdc_rx_cb() has to be invoked when copying file to msc. cdc & msc are 2 separated classes.

You are correct, i was meant to test CDC functionality, i am closing this issue.

closed 2022-07-25 by DNedic