hathach/tinyusb #1585: stm32fs_dev with stm32l4x2/stm32l4x3

HaimYosefHaim, 2022-08-01:
### Operating System

Linux

### Board

stm32L4x2

### Firmware

Compilation to a stm32l4x2.

### What happened ?

I tried to compile stm32fs_dev driver for stm32l4x2, I configured CFG_TUSB_MCU to OPT_MCU_STM32L4 as needed, but the linkage step failed.
When diving into stm32fs_dev, it seems like the stm32l4x2/stm32l4x3 are missing there, although it says in the description they are supported. 

### How to reproduce ?

1. add `#define STM32L432xx`
2. run the `cdc_msc` example while doing the necessary adjustments to `hw/bsp/stm32/l4/family.c` so the compilation will work. The interrupt handler is different and so is the `CLK_ENABLE_()` func
3. the linkage process will fail, because the functions implemented in dcd_stm32_fsdev were "defined" out during compilation/

what am I missing?

pigrew, 2022-08-01:
You likely got the error in `dcd_stm32_fsdev_pvt_st.h` stating:

```
#error You are using an untested or unimplemented STM32 variant. Please update the driver.
```

Please edit the header file and test. the contents should be something like:

```
#elif CFG_TUSB_MCU == OPT_MCU_STM32L4
  #include "stm32l4xx.h"
  #define PMA_LENGTH (1024u)
```

Also, add the L4 to the check in dcd_stm32_fsdev.c:

```
#if CFG_TUD_ENABLED && \
      ( TU_CHECK_MCU(OPT_MCU_STM32F0, OPT_MCU_STM32F3, OPT_MCU_STM32L0, OPT_MCU_STM32L1, OPT_MCU_STM32L4, OPT_MCU_STM32G4, OPT_MCU_STM32WB) || \
```

If you confirm this works, I can create the PR. Since only particular L4 models use the FSDEV device, the above logic will need to be slightly more complicated to select only them.

terjr, 2022-11-02:
I've been using tinyUSB on the STM32L452 successfully for a few months, so I can confirm it works with the following changes:

In file `dcd_stm32_fsdev.c`:
```
+#if defined(STM32L452xx)
+#define STM32L4_FSDEV
+#endif
```
```
#if TUSB_OPT_DEVICE_ENABLED && \
      ( TU_CHECK_MCU(OPT_MCU_STM32F0, OPT_MCU_STM32F3, OPT_MCU_STM32L0, OPT_MCU_STM32L1, OPT_MCU_STM32G4) || \
-        (TU_CHECK_MCU(OPT_MCU_STM32F1) && defined(STM32F1_FSDEV)) \
+        (TU_CHECK_MCU(OPT_MCU_STM32F1) && defined(STM32F1_FSDEV) || defined(STM32L4_FSDEV)) \
      )
```

Plus this in `dcd_int_enable` and `dcd_int_disable`
```
-#if CFG_TUSB_MCU == OPT_MCU_STM32F0 || CFG_TUSB_MCU == OPT_MCU_STM32L0
+#if CFG_TUSB_MCU == OPT_MCU_STM32F0 || CFG_TUSB_MCU == OPT_MCU_STM32L0 || CFG_TUSB_MCU == OPT_MCU_STM32L4
```

In file `dcd_stm32_fsdev_pvt_st.h`:
```
+#elif CFG_TUSB_MCU == OPT_MCU_STM32L4
+  #include "stm32l4xx.h"
+  #define PMA_LENGTH (1024u)
```

These changes works on top of tinyUSB v0.13.0, but I expect it to be similar now. I'm happy to test if you make a PR!

terjr, 2023-03-03:
This seems to be fixed in 0.15.0.

hathach, 2024-03-25:
thanks for confirmation, fixed and closed now

closed 2024-03-25 by hathach