hathach/tinyusb #1612: "multiple definition of `dcd_..." if cmpiling for STM32F2

DaGo1120, 2022-08-23:
### Operating System

Linux

### Board

Custom

### Firmware

`/*
 * tusb_config.h
 *
 *  Created on: Aug 23, 2022
 *      Author: DaGo1120
 */

#ifndef TUSB_CONFIG_H_
#define TUSB_CONFIG_H_

#ifdef __cplusplus
 extern "C" {
#endif

//--------------------------------------------------------------------+
// Board Specific Configuration
//--------------------------------------------------------------------+

// RHPort number used for device can be defined by board.mk, default to port 0
#ifndef BOARD_TUD_RHPORT
#define BOARD_TUD_RHPORT      		0
#endif

#ifndef CFG_TUSB_RHPORT0_MODE
#define CFG_TUSB_RHPORT0_MODE       (OPT_MODE_DEVICE | OPT_MODE_FULL_SPEED)
#endif

#ifndef CFG_TUSB_RHPORT1_MODE
#define CFG_TUSB_RHPORT1_MODE       0
#endif

// RHPort max operational speed can defined by board.mk
#ifndef BOARD_TUD_MAX_SPEED
#define BOARD_TUD_MAX_SPEED   		OPT_MODE_DEFAULT_SPEED
#endif

#ifndef BOARD_DEVICE_RHPORT_SPEED
#define BOARD_DEVICE_RHPORT_SPEED   OPT_MODE_FULL_SPEED
#endif

//--------------------------------------------------------------------
// Common Configuration
//--------------------------------------------------------------------

// defined by compiler flags for flexibility
#ifndef CFG_TUSB_MCU
#define CFG_TUSB_MCU          OPT_MCU_STM32F2
#endif

#ifndef CFG_TUSB_OS
#define CFG_TUSB_OS           OPT_OS_FREERTOS
#endif

#ifndef CFG_TUSB_DEBUG
#define CFG_TUSB_DEBUG        0
#endif

// Enable Device stack
#define CFG_TUD_ENABLED       1

[... 68 lines cut ...]
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_edpt_xfer':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:740: multiple definition of `dcd_edpt_xfer'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:733: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_remote_wakeup':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:575: multiple definition of `dcd_remote_wakeup'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:572: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_sof_enable':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:609: multiple definition of `dcd_sof_enable'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:606: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_edpt_open':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:631: multiple definition of `dcd_edpt_open'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:618: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_edpt_close_all':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:717: multiple definition of `dcd_edpt_close_all'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:706: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_edpt_stall':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:897: multiple definition of `dcd_edpt_stall'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:877: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_edpt_clear_stall':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:902: multiple definition of `dcd_edpt_clear_stall'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:882: first defined here
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: ./tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.o: in function `dcd_int_handler':
/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:1219: multiple definition of `dcd_int_handler'; ./tinyusb/src/portable/st/synopsys/dcd_synopsys.o:/home/danielg/STM32-projects/Testadapter_CAN-USB2V2/Debug/../tinyusb/src/portable/st/synopsys/dcd_synopsys.c:1128: first defined here
collect2: error: ld returned 1 exit status
make: *** [makefile:114: Testadapter_CAN-USB2V2.elf] Error 1
"make -j4 all" terminated with exit code 2. Build might be incomplete.

08:53:21 Build Failed. 25 errors, 0 warnings. (took 804ms)`

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-08-23:
mentioned in #1591, remove or filter the old driver `portable/st/synopsys/dcd_synopsys.c`. It will be removed from repo in the future soon.

DaGo1120, 2022-08-23:
This is the workaround I have been working with so far. Thanks for clarification!

closed 2023-03-20 by hathach