hathach/tinyusb #1617: error: "CFG_TUSB_DEBUG" redefined [-Werror] when LOG=1 on RP2040

todbot, 2022-08-28:
### Operating System

MacOS

### Board

Raspberry Pi Pico RP2040

### Firmware

tinyusb/examples/device/cdc_msc
using pico-sdk 1.4.0 in $PICO_SDK_PATH

### What happened ?

Cannot compile with `LOG=1` or `LOG=2`, instead get error:

```
[ 23%] Building C object CMakeFiles/cdc_msc.dir/src/main.c.obj
<command-line>: error: "CFG_TUSB_DEBUG" redefined [-Werror]
```

### How to reproduce ?

```
git clone https://github.com/raspberrypi/pico-sdk
git clone https://github.com/hathach/tinyusb/
export PICO_SDK_PATH=${PWD}/pico-sdk
cd tinyusb
git submodule update --init lib
cd examples/device/cdc_msc
make BOARD=raspberry_pi_pico get-deps
make BOARD=raspberry_pi_pico all

# the above compile works fine. the below fails
make BOARD=raspberry_pi_pico LOG=1 all
```

### Debug Log as txt file

Not debug log but part of the compile messages:

```
Scanning dependencies of target cdc_msc
make[3]: Leaving directory '/Users/tod/projects/tinyusb/tinyusb/examples/device/cdc_msc/_build/raspberry_pi_pico'
make[3]: Entering directory '/Users/tod/projects/tinyusb/tinyusb/examples/device/cdc_msc/_build/raspberry_pi_pico'
[ 23%] Building C object CMakeFiles/cdc_msc.dir/src/main.c.obj
<command-line>: error: "CFG_TUSB_DEBUG" redefined [-Werror]
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/cdc_msc.dir/build.make:76: CMakeFiles/cdc_msc.dir/src/main.c.obj] Error 1
make[3]: Leaving directory '/Users/tod/projects/tinyusb/tinyusb/examples/device/cdc_msc/_build/raspberry_pi_pico'
make[2]: *** [CMakeFiles/Makefile2:1465: CMakeFiles/cdc_msc.dir/all] Error 2
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-09-08:
indeed, the rp2040 cmake try to redefine this in two places, please manually comment out one in this line https://github.com/hathach/tinyusb/blob/master/hw/bsp/rp2040/family.cmake#L58 . I will make an PR to remove this later on.

todbot, 2022-09-08:
Thanks! I found that too, but I don't yet understand cmake enough to understand if making that change messed other things up, else I would make the PR too.

hathach, 2022-09-08:
I don't understand cmake as well, we can make an PR then mention rpi team to help. At worst, I guess we can temporarily comment out that line :)