hathach/tinyusb #1618: My successes and failures with TinyUSB (DWC2 mostly)

Jasoroony, 2022-08-29:
### Operating System

Windows 7

### Board

Various

### Firmware

Device CDC

### What happened ?

I've tried 5 different ways to use TinyUSB to produce a CDC device and these are my varied results:
1. ESP32-S2 (DWC2) - fails on a stress test (described in the ESP32 github) with a random IN packet (tx from device) not signalling an int. which halts that part of the stack.
2. Pi Pico native USB (unknown) - works perfectly, stress test as described passes.
3. Raspberry Pi Zero W (DWC2) - works perfectly, stress test as described passes.  (Bare metal implementation.)
4. Raspberry Pi 4B (DWC2) - fails on stress test.  Same issue as number 1, eventually a tx IN pk fails to generate an interrupt.
5. Pi Pico PIO-USB - fails to do much at 120MHz.  You can see attempts to set up the device but it always fails.  It does mount the device at 240MHz.  It fails to transmit and receive 64 byte packets reliably after that (but I note from searching the web, there are known issues with timing in PIO device mode - but I thought I'd throw it in since I went to the trouble of setting it all up..  It's also hard to debug, adding just about any debug output makes the timing worse and fail sooner.

So I got 2 out of 5 working (and in my mind, that makes TinyUSB pretty cool, and I can't see where there's a bug anywhere).

The DWC2 issue seems to be concerned with (from the device's perspective) a rx-tx-rx, sequence, however the host does a tx-tx sequence and since it's the one in control...  This corrupts the device's tx fifo somehow, seems to happen after the device's tx empty interrupt has loaded the data, just prior to getting the news it needs to accept data instead.  I find it interesting that the exact same code is being run on the Pi Zero W and 4B, however the 4B is the only one having issues...  I say that, except you're not really meant to use the 4B's native USB port, as it had an upgraded one added that does USB3.0.  I'll take a guess, they decided they didn't need to correct any errata in the chip due to that.

I reckon those are my final thoughts on it all too, other than the Pico and Pi Zero are both pretty cool.

### How to reproduce ?

It started before this issue, but that was closed - this is the one that remains open

https://github.com/espressif/arduino-esp32/issues/6221

### Debug Log as txt file

Nothing to log, this has already been repeated on the ESP32 github.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Jasoroony, 2022-09-03:
Bad news, I'm sorry to say...  Test number 2 above (Pi Pico native USB) with the loop part of my stress test moved from CPU0 to CPU1 (code moved from loop() to loop1() with a new small delay in start1() for coordination, but no other changes) cannot reliably transfer data either.

It all seemed to work OK at first, and while it was running I reviewed the Arduino code for Pico - seems to have extra mutexes that I don't see elsewhere, so that looked very well done.  Then after about 10 minutes and one GB transferred it just stopped (my stress test counts bytes, so I didn't get back the expected amount of data regardless of _what_ it was so the host stops sending, but I'm unsure where this has gone wrong on the device side).

I reset all the cables and the device and started it up again.  140MB in, a few minutes later, and it just stopped again.  I put it back to the way it was originally and got past a GB without trouble...  I'm sure when I first tested this it was running for a lot longer too.

With all the mutexes protecting normal execution of code, I'm thinking it doesn't like what's going on while servicing the USB interrupt - CPU0 servicing the interrupt while CPU1 is writing USB registers to set up tx'ing some more data at the same time?

hathach, 2022-09-08:
thanks for detailing your usage with tinyusb. Though this is more like a discussion than a bug report which needs step & firmware to reproduce. Note: if you are using it with other platform than bare metal C such as arduino (esp32/pio), there may be issue with platform integration code (e.g mutex between cores) and may not entirely be the bug of the stack.