hathach/tinyusb #1626: tinyUSB CDC messages are repeated full speed infinitive

ralphde, 2022-09-06:
### Operating System

Others

### Board

ESP32-S2

### Firmware

ESP-IDF 4.4.2

### What happened ?

occasionally messages sent are repeated infinitive . sending a new message stops this or get repeated infinitive as well.

### How to reproduce ?

required a bigger application making use of stack and heap more exessive

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-09-13:
Do you have any simplest code and "exact" steps to reproduce the issue.

ralphde, 2022-09-13:
Nop, it happens occasionally in the ESP-IDF implementation and when it happens is a backtrace impossible due to muktitasking. It must be that tinyUSB keeps thinking that the transmit buffer is not sent and sends it again. I was hoping that here somebody knows tinyUSB to point me to the source where (as CDC device) the data buffer is handled. I think that a byte buffer goes out of bound and overwrites e.g. the flag marking the buffer as sent.

hathach, 2022-09-13:
I will move this to discussion, since there is no way to fix a bug without an way to reproduce it reliably. Also IDF also add its own integration/management code on top of tinyusb, you may need to also look at those as well. cdc_device.c is available at src/class/cdc you could check that out.