hathach/tinyusb #1634: STM32H730: wrong handling of 1 port

diggit, 2022-09-11:
### Operating System

Linux

### Board

Custom board

### Firmware

FW with custom RTOS. Not relevant for this issue.

### What happened ?

OOB access to `_dwc2_controller` array.

[Note](https://github.com/hathach/tinyusb/blob/8fe9022a6e6dea785092f4e0dff8180f1f0d4d86/src/portable/synopsys/dwc2/dwc2_stm32.h#L65) about STM32H73x having only one port is right.
```c
// NOTE: H7 with only 1 USB port: H72x / H73x / H7Ax / H7Bx
// USB_OTG_FS_PERIPH_BASE and OTG_FS_IRQn not defined
```

Thus [_dwc2_controller](https://github.com/hathach/tinyusb/blob/8fe9022a6e6dea785092f4e0dff8180f1f0d4d86/src/portable/synopsys/dwc2/dwc2_stm32.h#L94) has only one item, because `USB_OTG_FS_PERIPH_BASE` is not defined.
```c
// On STM32 for consistency we associate
// - Port0 to OTG_FS, and Port1 to OTG_HS
static const dwc2_controller_t _dwc2_controller[] =
{
#ifdef USB_OTG_FS_PERIPH_BASE
  { .reg_base = USB_OTG_FS_PERIPH_BASE, .irqnum = OTG_FS_IRQn, .ep_count = EP_MAX_FS, .ep_fifo_size = EP_FIFO_SIZE_FS },
#endif

#ifdef USB_OTG_HS_PERIPH_BASE
  { .reg_base = USB_OTG_HS_PERIPH_BASE, .irqnum = OTG_HS_IRQn, .ep_count = EP_MAX_HS, .ep_fifo_size = EP_FIFO_SIZE_HS },
#endif
};
```
From my understanding, TinyUSB convention is to mark HS port as number 1, thus define `CFG_TUSB_RHPORT1_MODE` in config, that results to `TUD_OPT_RHPORT` being 1 (at least that was convention in some old tusb_config.h file, I recycled in past).  `tud_init(...)` receives `TUD_OPT_RHPORT` as argument which is later used to deduce address of dwc2 regs from `_dwc2_controller` at position 1 which is OOB access.

What is correct solution here? Just use `CFG_TUSB_RHPORT0_MODE`?

I propose some `assert` guarding to prevent such issues in future.

### How to reproduce ?

1. Have device with just one USB port, which is HS (H72x / H73x / H7Ax / H7Bx)
2. use `CFG_TUSB_RHPORT1_MODE` for port configuration
3. get OOB access to `_dwc2_controller` array.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Nikitarc, 2022-09-11:
This is already  modified in v 14:

`#elif CFG_TUSB_MCU == OPT_MCU_STM32H7
  #include "stm32h7xx.h"
  #define EP_MAX_FS       9
  #define EP_FIFO_SIZE_FS 4096
  #define EP_MAX_HS       9
  #define EP_FIFO_SIZE_HS 4096
  #if (! defined USB2_OTG_FS)
 	// H7 with only 1 USB port: H72x / H73x / H7Ax / H7Bx
    // USB_OTG_FS_PERIPH_BASE and OTG_FS_IRQn not defined in CMSIS files
    #define USB_OTG_FS_PERIPH_BASE  USB1_OTG_HS_PERIPH_BASE
    #define OTG_FS_IRQn             OTG_HS_IRQn
  #endif`

hathach, 2022-09-13:
I have done some refactor, and probably mess up when merging/rebasing, will be fixed by #1641 . Thanks for the issue.

PR #1641 (2022-09-13): re-added code in pr 1438 to fix compile with h7 with 1 usb otg
fix #1634
closed 2022-09-13 by hathach