hathach/tinyusb #1658: Wrong usb488 subclass capabilities bitorder

DaGo1120, 2022-09-29:
### Operating System

Others

### Board

doesn't matter

### Firmware

Patch: https://paste.debian.net/1255382/

### What happened ?

The bitorder in the usb488 subclass capabilities  is wrong. I tried to attach a Patch, but its format gets corrupted, even if i use the "code" formatter. So i pasted it to paste.debian.net, see above.

### How to reproduce ?

.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

DaGo1120, 2022-09-30:
I had to edit the report with an updated link to paste.debian.net (just in case you missed the patch). The new pate expires after 90 days.

fenugrec, 2022-10-07:
> The bitorder in the usb488 subclass capabilities is wrong. I tried to attach a Patch, but its format gets corrupted, even if i use the "code" formatter. 

Seems to work fine here :

```
From ec1a30ded4f6c5580f99a6daa43d341f352920a0 Mon Sep 17 00:00:00 2001
From: Daniel Gorsulowski
Date: Thu, 29 Sep 2022 10:24:48 +0200
Subject: [PATCH] fixed usb488 subclass capabilities bitorder

See Test & Measurement Class Specification
https://www.usb.org/document-library/test-measurement-class-specification
---
 src/class/usbtmc/usbtmc.h | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/src/class/usbtmc/usbtmc.h b/src/class/usbtmc/usbtmc.h
index e7016ae24..c3483cab6 100644
--- a/src/class/usbtmc/usbtmc.h
+++ b/src/class/usbtmc/usbtmc.h
@@ -277,17 +277,17 @@ typedef struct TU_ATTR_PACKED
 
   struct TU_ATTR_PACKED
   {
-    unsigned int is488_2 :1;
-    unsigned int supportsREN_GTL_LLO :1;
     unsigned int supportsTrigger :1;
+    unsigned int supportsREN_GTL_LLO :1;
+    unsigned int is488_2 :1;
   } bmIntfcCapabilities488;
 
   struct TU_ATTR_PACKED
   {
-    unsigned int SCPI :1;
-    unsigned int SR1 :1;
-    unsigned int RL1 :1;
     unsigned int DT1 :1;
+    unsigned int RL1 :1;
+    unsigned int SR1 :1;
+    unsigned int SCPI :1;
   } bmDevCapabilities488;
   uint8_t _reserved3[8];
 } usbtmc_response_capabilities_488_t;
-- 
2.25.1
```

For multi-line code formatting it needs the triple-backquote - either insert them manually, or paste your text first, select it, then click the "Code" button

hathach, 2022-10-10:
ah right, the struct layout is incorrectly inverted. It would be great if you could make an PR, otherwise, please wait a bit, I will do it myself.

PR #1969 (2023-03-24): Fix tmc488 bit order 1658
**Describe the PR**
Fix #1658 bit field ordering normally start from 0 for first field

[image]
[image]
closed 2023-03-24 by hathach