hathach/tinyusb #1663: Keyboard Stalled

Dave-Osborne, 2022-10-05:
### Operating System

Others

### Board

Raspberry Pi Pico

### Firmware

Custom firmware based on examples.

### What happened ?

I have a specific keyboard which fails to mount and always reports 'Control STALLED' followed by process_set_config 428: ASSERT FAILED.

Other keyboards I have access to appear to work as expected.

### How to reproduce ?

Plug in keyboard.

### Debug Log as txt file

HID Set Protocol = 0
[0:1] Set Interface: 21 0B 00 00 00 00 00 00 
hw_endpoint_init dev 1 ep 0 out xfer 0
dev 1 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 0
hw_endpoint_init dev 1 ep 0 in xfer 0
dev 1 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000
Stall REC
on EP 80 with 0 bytes
[0:1] Control STALLED

process_set_config 428: ASSERT FAILED

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-10-10:
can you provide details on the keyboard in question
- specific name, model, product link
- descriptor of keyboard including configuration descriptor and hid report descriptor
- For log, please **attached** full txt file when LOG=2

Dave-Osborne, 2022-10-12:
Hi @hathach - the keyboard itself is an OEM product, this is the product link but it doesn't contain too much detail.

https://www.keyboardspecialists.co.uk/collections/kbskeyboards/products/kbs-pc-c?variant=9634038775850

It works ok with Windows and a PIC based USB host (which we are looking to replace with a TinyUSB powered Pico).

I've since also tried it with my MacBook which definitely doesn't like it (see attached), this now makes me think the keyboard isn't 100% USB compliant.

https://user-images.githubusercontent.com/19535389/195295386-957cf28c-19bc-435e-86c7-14e1631381ef.mp4

All of the debug output below when plugging it into the Pico below....thanks Dave.

[0:] USBH DEVICE ATTACH
hcd_port_reset
hcd_port_connect_status
Full Speed
[0:0] Open EP0 with Size = 8
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00 
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 8
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x7408  [1] = 0x0000
Buffer complete
buf_status 0x00000001
Single Buffered: ep_ctrl = A0000180
  Sync BufCtrl: [0] = 0xe008  [1] = 0x0000
Completed transfer of 8 bytes on ep 0 in
Transfer complete
on EP 80 with 8 bytes
[0:0] Control data:
  0000:  12 01 10 01 00 00 00 08                          |........|
hcd_edpt_xfer dev_addr 0, ep_addr 0x0, len 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0xf400  [1] = 0x0000
Buffer complete
buf_status 0x00000001
Single Buffered: ep_ctrl = A0000180
  Sync BufCtrl: [0] = 0x6000  [1] = 0x0000
  Short packet on buffer 0 with 0 bytes
Completed transfer of 0 bytes on ep 0 out
Transfer complete
on EP 00 with 0 bytes

Set Address = 1
[0:0] Set Address: 00 05 01 00 00 00 00 00 
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 0
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000
[... 192 lines cut ...]
Single Buffered: ep_ctrl = A0000180
  Sync BufCtrl: [0] = 0xe000  [1] = 0x0000
  Short packet on buffer 0 with 0 bytes
Completed transfer of 0 bytes on ep 0 in
Transfer complete
on EP 80 with 0 bytes

HID Set Protocol = 0
[0:1] Set Interface: 21 0B 00 00 00 00 00 00 
hw_endpoint_init dev 1 ep 0 out xfer 0
dev 1 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 8 bytes
hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 0
hw_endpoint_init dev 1 ep 0 in xfer 0
dev 1 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000
Stall REC
on EP 80 with 0 bytes
[0:1] Control STALLED

process_set_config 428: ASSERT FAILED

hathach, 2022-10-12:
please put the debug text in a LOG.TXT file

Dave-Osborne, 2022-10-12:
Sorry, updated with log as attachment.