hathach/tinyusb #1669: Can't find pio_ush.h (/hw/bsp/rp2040/family.c)

paulhamsh, 2022-10-09:
### Operating System

Others

### Board

Pico Pi

### Firmware

None - compiling on WSL in Windows.

### What happened ?

I'm trying to build host_hid_to_device_cdc in my own directory.
I get an error saying can't find pio_usb.h - which seems fair because I can't find it.

It is referred to in /hw/bsp/rp2040/family.c:

```
#if CFG_TUH_RPI_PIO_USB || CFG_TUD_RPI_PIO_USB
#include "pio_usb.h"
#endif
```

But if I ```cd /root/pico/pico-sdk``` and ``` ls -R | grep "pio_usb.h"``` I get nothing.

```
[ 86%] Building C object CMakeFiles/host_hid_to_device_cdc.dir/root/pico/pico-sdk/lib/tinyusb/src/tusb.c.obj
[ 87%] Building C object CMakeFiles/host_hid_to_device_cdc.dir/root/pico/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj
[ 88%] Building C object CMakeFiles/host_hid_to_device_cdc.dir/root/pico/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c.obj
/root/pico/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c:39:10: fatal error: pio_usb.h: No such file or directory
   39 | #include "pio_usb.h"
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/host_hid_to_device_cdc.dir/build.make:716: CMakeFiles/host_hid_to_device_cdc.dir/root/pico/pico-sdk/lib/tinyusb/hw/bsp/rp2040/family.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1624: CMakeFiles/host_hid_to_device_cdc.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

My CMakeLists.txt is: 

```
cmake_minimum_required(VERSION 3.13)
include(pico_sdk_import.cmake)
project(pico_host_pio C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

add_executable(host_hid_to_device_cdc)

# Example source
target_sources(host_hid_to_device_cdc PUBLIC
        ${CMAKE_CURRENT_LIST_DIR}/tusb_config.h
        ${CMAKE_CURRENT_LIST_DIR}/main.c
        ${CMAKE_CURRENT_LIST_DIR}/usb_descriptors.c
        )

# Make sure TinyUSB can find tusb_config.h
target_include_directories(host_hid_to_device_cdc PUBLIC
        /root/pico/pico-sdk/lib/tinyusb
        ${CMAKE_CURRENT_LIST_DIR})

# In addition to pico_stdlib required for common PicoSDK functionality, add dependency on tinyusb_host
# for TinyUSB device support and tinyusb_board for the additional board support library used by the example
target_link_libraries(host_hid_to_device_cdc PUBLIC pico_stdlib tinyusb_host tinyusb_board)

pico_add_extra_outputs(host_hid_to_device_cdc)

```

### How to reproduce ?

```
mkdir tiny
cd tiny
git clone https://github.com/hathach/tinyusb
ls -R | grep "pio_usb"

pio_usb
./tinyusb/src/portable/raspberrypi/pio_usb:
dcd_pio_usb.c
hcd_pio_usb.c

ls -R | grep "pio_usb.h"
// NOTHING 

cat tinyusb/hw/bsp/rp2040/family.c | grep "pio_usb.h"
#include "pio_usb.h"
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

paulhamsh, 2022-10-09:
Ok I found this site which took me a step closer:

https://github.com/bgni/minimal-pico-tinyusb-pio-project

I need to 

```
cd ~/pico/pico-sdk/lib/tinyusb

git -C tinyusb submodule update --init hw/mcu/raspberry_pi/Pico-PIO-USB
```

Then the file now exists - not really sure why it wasn't 'somewhere' before, but this bit of magic made it appear.

Now the 'can't find' error moved to hcd_pio_usb.c so perhaps it is now just because I need to set a directory in the cmake stuff.....
Will keep trying

paulhamsh, 2022-10-09:
I've sorted it - but not sure how anyone is supposed to know this stuff?

closed 2022-10-09 by paulhamsh

noisysoil, 2022-11-14:
@paulhamsh Perhaps you can re-open this bug for investigation - I've encountered the same problem.   IMHO the pio_usb.h file should be included in the source from release 0.14.0.

paulhamsh, 2022-11-14:
@noisysoil  did you try this:

cd ~/pico/pico-sdk/lib/tinyusb

git -C tinyusb submodule update --init hw/mcu/raspberry_pi/Pico-PIO-USB

noisysoil, 2022-11-14:
> cd ~/pico/pico-sdk/lib/tinyusb

A little correction - for the submodule update this should be `cd ~/pico/pico-sdk/lib` (it's also best to reference `$PICO_SDK_PATH` environment variable instead of your directory).

However, `git -C tinyusb submodule update --init hw/mcu/raspberry_pi/Pico-PIO-USB` submodule update didn't work (same issue not finding `pio_usb.h`).

However, after some head-scratching, I performed a `git submodule update --init --recursive` in the Pico SDK source root (`$PICO_SDK_PATH`); and compiled with the following in `tusb_config.h`:
```
#if CFG_TUSB_MCU == OPT_MCU_RP2040
#define CFG_TUH_RPI_PIO_USB   1
#endif
```
I didn't add any `Pico-PIO-USB/src/` etc in the Makefile etc, and it compiled and ran fine a standard tinyUSB example file and my custom gamepad one.

Does this work for you?

noisysoil, 2022-11-14:
@paulhamsh Forgot to mention for completeness the above is on Pico SDK tag 1.4.0.

paulhamsh, 2022-11-14:
Good point - I can’t remember if the submodule I ran actually worked.
The sdk root one definitely did.
But what I can’t work out is how to get ONLY the pico pi submodules and not them all - and there are a lot!

paulhamsh, 2022-11-14:
This is what I eventually did - shown here: 

```https://github.com/paulhamsh/PicoPiUSBMidiHostDevice```

But I'm going to try the 

```CFG_TUH_RPI_PIO_USB   ```

noisysoil, 2022-11-22:
@paulhamsh Apologies for the delay, here's my CMakeLists.txt for the project (with some cruft removed):

```
cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

set(FAMILY rp2040)
set(BOARD pico_sdk)

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)
if (PICO_SDK_VERSION_STRING VERSION_LESS "1.4.0")
    message(FATAL_ERROR "Raspberry Pi Pico SDK version 1.4.0 (or later) required. Your version is ${PICO_SDK_VERSION_STRING}")
endif()

project(joystick C CXX ASM)

set(PICO_COPY_TO_RAM 1)
set(CFG_TUSB_DEBUG 0)

# Creates a pico-sdk subdirectory in our project for the libraries
pico_sdk_init()

add_executable(${PROJECT_NAME})

target_sources(${PROJECT_NAME} PUBLIC
        ${CMAKE_CURRENT_SOURCE_DIR}/src/main.cpp
        ${CMAKE_CURRENT_SOURCE_DIR}/src/usb_descriptors.c)
target_include_directories(${PROJECT_NAME} PUBLIC
        ${CMAKE_CURRENT_SOURCE_DIR}/src/include/
        )

target_link_libraries(${PROJECT_NAME} PUBLIC pico_stdlib tinyusb_device tinyusb_board)

# Disable usb output, disable uart output as TinyUSB's CDC is being used instead.
pico_enable_stdio_usb(${PROJECT_NAME} 0)
pico_enable_stdio_uart(${PROJECT_NAME} 0)

pico_set_program_name(${PROJECT_NAME} "joystick")
pico_set_program_description(${PROJECT_NAME} "joystick")
pico_set_program_version(${PROJECT_NAME} "0.9")
pico_set_program_url(${PROJECT_NAME} "")

pico_add_extra_outputs(${PROJECT_NAME})
```

And in the main code relevant includes are:
```
#include "tusb.h"
#include "pico.h"
#include "pico/stdlib.h"
```

I've found that when making big changes to `CMakeLists.txt` the CMake cache needs to be blitzed, and sometimes the `build` directory too.