hathach/tinyusb #1673: Support two physcial host ports on the RP2040 (native and pio)

hathach, 2022-10-10:
### Discussed in https://github.com/hathach/tinyusb/discussions/1631

<div type='discussions-op-text'>

<sup>Originally posted by **morio** September 10, 2022</sup>
Is it possible to use TinyUsb to control two physical USB host ports?
Specifically on the RP2040 I'd like to use the builtin USB as host and the Pico PIO USB also as a host port at the same time.
This would be to allow a USB keyboard in one port and a USB mouse in the other. 

I've seen the host_hid_to_device example in the Pico PIO USB project (https://github.com/sekigon-gonnoc/Pico-PIO-USB/tree/main/examples/host_hid_to_device_cdc) but that uses a device and a host.

Thanks!</div>

webbbn, 2022-10-13:
> I will also appreciate that TinyUSB can work with 2 physicals devices ports to permit to link 2 usb host systems.

I inquired about this a while back, for other devices that support multiple device interfaces (STM32, etc), and it was not considered a priority.

Maybe there's some issues that I don't understand with doing that, but, IMO, it would be more useful than dual host.

My use case is in emulating a device from a machine that doesn't support device mode, like a standard PC. AFIK, there is no way to support that application without some device in between that support device mode to the PC and device mode to the other device. The only solution that I've found is to connect a pair of microcontroller boards together via UART, or some such.

Dual host would be nice, but that can be solved using a standard hub.

cleverca22, 2023-07-31:
> Dual host would be nice, but that can be solved using a standard hub.

but that wont cover a case like the rpi4(00), where it has 2 host controllers, routed to different places

there is the pcie/vl805/xhci controller, going to the main 4 usb-a ports
and there is either dwc or xhci, going to the usb-c port

if you limit yourself to one xhci controller, you have to choose which physical ports work
if you limit yourself to an xhci driver that can handle 2 controllers, you cant use device mode
if you support both dwc2 and xhci drivers at once, you can freely flip the usb-c between host/device, while still using the usb-a's

but to properly support that, i think you would need to replace all of the global `hcd_*()` functions with a table of function pointers
and then route the call to the right one for the root port
but i can see how an abstraction layer could be created
root port 0-3 go to the vl805/xhci, root port 4 goes to the dwc

eivindbohler, 2023-09-21:
I've started to look at how I can get both the native USB host and the Pico PIO USB host up and running at the same time on the RP2040. My use case is as a HID controller for keyboard + mouse on a retro ITX motherboard I'm making (https://www.vogons.org/viewtopic.php?t=93480). I find it quite challenging however, because of how mutually exclusive the two solutions are set up to be for the RP2040 specifically, and generally how TinyUSB doesn't yet support two host controllers. In also not that familiar with this project, so I have to learn as I go along.

@hathach, what's your take on this? Do you feel it makes the most sense to try to get the Pico PIO host working "alongside" and as decoupled from the native host as possible, or put in place a more general support for multiple host controllers? I'd imagine the latter being more elegant and potentially useful for other boards as well, but I'm sure harder to achieve.

I'm open to suggestions and tips on how to best tackle this. Thx!

tannewt, 2023-09-21:
@eivindbohler Why do you need two host controllers? A hub would allow you to connect multiple devices to one host port.

eivindbohler, 2023-09-21:
@tannewt Because _not_ having to rely on a hub is arguably the better solution. I mean, if this can be done in software then why not?
As for hubs, I assume most external ones would probably work just fine - but the fewer external dependencies the better. I could also use an internal hub chip I guess. The problem with that is that it would break being able to reprogram the rp2040 over the native USB port (using the BOOTSEL button/signal).

At a high level - both the rp2040 and the TinyUSB library are extremely capable and have more than enough juice to handle the host HID and PS/2 signal generation I need - and I've successfully managed to run this both in native and Pico PIO mode, just not both simultaneously (yet).

hathach, 2023-09-26:
> @hathach, what's your take on this? Do you feel it makes the most sense to try to get the Pico PIO host working "alongside" and as decoupled from the native host as possible, or put in place a more [...]

native usb controller + pio-usb host can coexist, treated as mixed host controllers. There are more MCUs with mixed controllers lately, the approach I would like to do is wrapping all the APIs hcd.h into an struct of pointers (similar to how the class drivers are done), and each controller will declare its own variable holding these APIs.

horsepork, 2024-01-15:
> PS: pio-usb can actually add more usb host ports, I haven't tested that but it is possible to have multiple hub port with pio-usb host

How would this be done? Is there an example somewhere?