hathach/tinyusb #1677: CDC Device unable to transfer large amounts of data

qwertymodo, 2022-10-12:
### Operating System

Windows 10

### Board

Raspberry Pi Pico

### Firmware

examples/device/cdc_msc

### What happened ?

I am attempting to utilize the CDC device class to stream a large amount of data to the host PC, for example, to implement an EEPROM reader.  However, I am unable to transfer more than about 128 bytes before the tud_cdc_write function simply stops sending any more data.  The only way I have been able to get around this is by adding hard-coded delays between calls to tud_cdc_write.  It seems like the correct solution is to use tud_cdc_write_available to determine whether or not it is possible to send new data, but in my testing, once it stops sending data, tud_cdc_write_available only ever returns 0.  It's hard to tell if this is a bug or simply user error, because none of the existing examples show how to use the fifo status flags for gating large transfers, and the relatively large delays inherent in the existing terminal echo examples (i.e. several ms between user keystrokes) eliminates the need for such checks.

### How to reproduce ?

Start with examples/device/cdc_msc, comment out the call to led_blinking_task() in main, and replace the void cdc_task(void) function with the following:

```
void cdc_task(void)
{
  char buf[64];
  
  if (tud_cdc_connected() )
  {
    char avail = tud_cdc_write_available();

    if ( avail )
    {
      led_blinking_task();
      for (int i = 0; i < 64; ++i)
      {
        buf[i] = i;
      }
      tud_cdc_write(buf, avail);
      tud_cdc_write_flush();
    }
  }
}
```

Attach a serial terminal emulator to the CDC device.

Expected behavior would be for the board LED to blink, and for the serial terminal to constantly be receiving bytes from the value 0 up to however many bytes were available to send on that particular loop (ideally 0 to 63, if the device was able to transmit fast enough to keep up).  However, instead, no data is received, and the LED does not blink, indicating that tud_cdc_write_available is always returning 0.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-10-13:
change type of avail to uint32_t that matches the return type of tud_cdc_write_available(). It probably return 256 which is 0 in char type.

qwertymodo, 2022-10-13:
I tried that, it doesn't seem to work.  I even tried copying [this example](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb.c#L63) from the Raspberry Pi Pico SDK stdio_usb.c code, and this just crashes my serial terminal when I try to connect to the port, and still no blinking of the board LED.
  
```
void cdc_task(void)
{
  char buf[SERIAL_BUFFER_SIZE];

  for (int i = 0; i < SERIAL_BUFFER_SIZE; ++i)
  {
    buf[i] = i;
  }

  if (tud_cdc_connected() )
  {
    led_blinking_task();
    for (int i = 0; i < SERIAL_BUFFER_SIZE;)
    {
      int n = SERIAL_BUFFER_SIZE - i;
      int avail = (int) tud_cdc_write_available();
      if (n > avail) n = avail;
      if (n)
      {
        int n2 = (int) tud_cdc_write(buf + i, (uint32_t)n);
        tud_task();
        tud_cdc_write_flush();
        i += n2;
      }
    }
  }
```

htalaco, 2023-11-27:
I have the exact same problem

HiFiPhile, 2026-10-31:
As found later in this thread, the stall comes from the loop not calling `tud_task()` while waiting on `tud_cdc_write_available()`: the stack never processes the transfer completion, so the FIFO never drains. Call `tud_task()` inside the wait loop, or return to the main loop. Closing.

closed 2026-10-31 by HiFiPhile