hathach/tinyusb #1684: ST Synopsys port lacks SOF support

battlesnake, 2022-10-15:
### Related area

ST Synopsys port

### Hardware specification

Cortex M7, STM32H750, USB3300, custom board

### Is your feature request related to a problem?

The port currently lacks support for USB SOF frames.

### Describe the solution you'd like

I will provide a PR with my proposed solution

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

battlesnake, 2022-10-15:
PR: https://github.com/hathach/tinyusb/pull/1685

battlesnake, 2022-10-19:
I didn't use try it.  I didn't find info in the docs about which port(s) to use, so I tried `st/synopsys` (since using ST part), and it worked.

battlesnake, 2022-10-19:
Besides (presumably) supporting SOF already, what are other differences of that synopsys/dw2 port vs. st/synopsys?

PanRe, 2022-10-20:
It fuses the device driver for multiple MCUs as far as i know. I don't know why the `st/synopsis` driver is still there, maybe due to backward compatibilities.

battlesnake, 2022-10-20:
Thanks, I'll give it a try in an hour or so.

battlesnake, 2022-10-22:
Using synopsis/dwc2, I get into a crash-loop.
Almost certainly due to configuration on my side though.

I'll go back to using st/synopsys for now.
Once I've got my UAC2 interrupt status endpoint working, then I'll try synopsys/dwc2 again and see if I can either fix the issue in my code, or file a PR if it's due to a bug in tinyusb.