hathach/tinyusb #1697: Interrupt-endpoint for audio class

battlesnake, 2022-10-20:
### Related area

Audio class UAC2

### Hardware specification

STM32H7 (probably not relevant)

### Is your feature request related to a problem?

UAC2 supports an interrupt endpoint to provide notifications to the hose about control changes (e.g. mute button pressed on the device).

I see that there is some code for this, but it seems incomplete.
I see `audio_device.c` never assigns a value to `audio->ep_int_ctr`.
Compare to feedback endpoint (which I have working), which assigns and clears `audio->ep_fb` as needed.

### Describe the solution you'd like

Support for the interrupt endpoint in Audio Class devices.

I can implement this myself, but any advice/guidance beforehand would be appreciated!

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

battlesnake, 2022-10-23:
Working, but possibly ugly implementation: https://github.com/hathach/tinyusb/pull/1702

Gadgetoid, 2024-03-12:
I came across the linked PR while searching for a means to synchronise device control changes (volume and mute in particular) to the host. I'd like to register my interest in this feature.

battlesnake, 2024-03-15:
Sure.  My fork of tinyusb has a few extra features implemented for audio class, but didn't get merged as I don't have the boards used by the mainline project for testing.
If you don't need any newer features than my fork (or official support from the project), you could potentially use my fork.

Gadgetoid, 2024-03-22:
> If you don't need any newer features than my fork (or official support from the project), you could potentially use my fork.

Unfortunately I think I need both, I'd hoped I could get away with cherry-picking some bits from your fork but so far it's mostly lead to eclectic bugs and crashes.

Thanks for blazing this trail, anyway, there are surely others out there like me who think using media keys to have the host control the volume is barbaric 😆

closed 2024-09-08 by battlesnake