hathach/tinyusb #1705: Mouse poll rate cannot go over 500Hz on MacBook with Apple Silicon

ViRb3, 2022-10-30:
### Operating System

MacOS

### Board

Raspberry Pico

### Firmware

usb/device/dev_hid_composite

### What happened ?

I am trying to simulate a mouse moving at 1000Hz poll rate with the Raspberry Pi Pico. With the [default descriptors](https://github.com/hathach/tinyusb/blob/master/examples/device/hid_composite/src/usb_descriptors.c), I am getting a polling rate of 250Hz. I changed the polling interval from `5` to `1` and I got 1000Hz on my MacBook 2019 with Intel CPU, as well as my Windows laptop, but only 500Hz on my two MacBook 2020 with Apple CPU:

https://github.com/hathach/tinyusb/blob/154821acc02f8c7917ed7e94318de0a26f883b75/examples/device/hid_composite/src/usb_descriptors.c#L114

This leads me to believe that the problem is in the descriptors that TinyUSB uses by default. I verify the polling rate using this website:
- https://devicetests.com/mouse-rate-test

With my gaming mouse, it reports 1000Hz. With my cheap mouse, it reports 100Hz. I was also able to simulate 1000Hz using CircuitPython by applying this patch:
- https://github.com/ViRb3/circuitpython/commit/3cee549a5a0fd82cc2c70d9d068a77e38d298697

CircuitPython code, 1000Hz:

```python
import usb_hid

mouse = usb_hid.Device.MOUSE
mouse_move_report = bytearray(4)

mouse_move_report[1] = 1
mouse_move_report[2] = 1

while True:
    mouse.send_report(mouse_move_report)
```

TinyUSB code, 500Hz:

```c
int main(void)
{
  board_init();
  tusb_init();

  while (1)
  {
    tud_task();
    // skip if hid is not ready yet
    if ( !tud_hid_ready() ) continue;
    tud_hid_mouse_report(REPORT_ID_MOUSE, 0, 1, 1, 0, 0);
  }

  return 0;
}
```

I figured this might be a configuration issue, so to the best of my abilities I cloned the following properties from CircuitPython:
- HID Report Descriptor
- USB Device Descriptor
- Removed horizontal scroll from report (maybe extra size slows it down?)

Sadly, none of this changes anything. Here are my sources from CircuitPython:
- https://github.com/adafruit/circuitpython/blob/main/shared-module/usb_hid/__init__.c
- https://github.com/adafruit/circuitpython/blob/main/shared-module/usb_hid/Device.c
- https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/usb/usb_desc.c

Thanks!

### How to reproduce ?

```c
int main(void)
{
  board_init();
  tusb_init();

  while (1)
  {
    tud_task();
    // skip if hid is not ready yet
    if ( !tud_hid_ready() ) continue;
    tud_hid_mouse_report(REPORT_ID_MOUSE, 0, 1, 1, 0, 0);
  }

  return 0;
}
```

Test in:
- https://devicetests.com/mouse-rate-test

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

sonodima, 2024-08-04:
I am currently experiencing the same issue.
By any chance, did you find what was wrong with it?

ViRb3, 2024-08-08:
Unfortunately no, had to pause my project until a solution is found. Still hoping that happens one day.

sonodima, 2024-09-11:
I tested Adafruit's TinyUSB abstraction for Arduino and it can reach 1kHz with the RP2040.

For now (unfortunately) I think I'll have settle with that for my project.
It might be worth it to take a look at their implementation though.

sonodima, 2024-09-12:
After further testing, I noticed that disabling the USB CDC Serial (which is enabled by default) in the Adafruit TinyUSB library results in the polling rate to get limited to 500Hz on macOS.
On Windows it's still 1KHz, and that is consistent with the problems we had before using TinyUSB directly.

If the USB CDC Serial is enabled, it has zero problems reaching the full 1KHz on both macOS and Windows.