hathach/tinyusb #1715: Add fuzz testing harnesses

silvergasp, 2022-11-02:
### Related area

Fuzz testing

### Hardware specification

Host only

### Is your feature request related to a problem?

No

### Describe the solution you'd like

I'd like to improve the security and robustness of the tinyusb stack by adding a set of fuzz tests.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

silvergasp, 2022-11-02:
I'm happy to spend some time contributing to this one. I've got some general questions first though;
1. Is it acceptable to you to have the fuzz harnesses written in a combo of c/c++. These fuzz harnesses would be run from the host only, so I think the overhead of c++ is acceptable given the ease of use of the [fuzzing libs available in c++](https://github.com/llvm/llvm-project/blob/main/compiler-rt/include/fuzzer/FuzzedDataProvider.h).
2. Do you have the bandwidth to give some guidance on WIP PRs?

Here is a list of the things that I would like to add;
- [ ] Add build infrastructure for fuzz tests. (WIP) #1716
- [ ] Add a fuzz harness for each class.
    - [ ] audio
    - [ ] bth
    - [ ] cdc (WIP) #1716
    - [ ] dfu
    - [ ] hid
    - [ ] midi
    - [ ] msc
    - [ ] net
    - [ ] usbtmc
    - [ ] vendor
    - [ ] video
- [ ] Add CI configuration to continuously fuzz.
- [ ] Add dynamic configs for composite class (e.g. cdc+msc, or cdc+cdc+audio etc.).

silvergasp, 2022-11-07:
I'd like to add an additional question. Would you be open to using google's OSS-fuzz service for continuous fuzzing? It's a free service and will fuzz and send you updates when bugs are found.

https://github.com/google/oss-fuzz

hathach, 2022-12-05:
Awesome, thank you very much for spending time and effort to adding fuzzing support. 

> 1. Is it acceptable to you to have the fuzz harnesses written in a combo of c/c++. These fuzz harnesses would be run from the host only, so I think the overhead of c++ is acceptable given the ease [...]

Sure thing, I have no problems with C/C++ combo at all as long as it is easy to use/maintain. 

> 2. Do you have the bandwidth to give some guidance on WIP PRs?

Sure things, though sometime I have to do lots of work switching and catching up due to other works. But I absolutely have time for this. 

> I'd like to add an additional question. Would you be open to using google's OSS-fuzz service for continuous fuzzing? It's a free service and will fuzz and send you updates when bugs are found.
> 
> https://github.com/google/oss-fuzz

Yeah sure, why not, I am happy if we could be ale to use this service. Though I am new to fuzzing and still learn on how to use/run/test with it.

Thank you again for this super helpful issue and sorry for super late response :)

silvergasp, 2022-12-06:
Sweet, I'll open a WIP progress PR over at OSS-fuzz to integrate tinyusb. Is your '@tinyusb.org` email address associated with a google account? I'm just going over the requirements for accepting new projects with OSS-fuzz and one of them is a google email address.

https://google.github.io/oss-fuzz/getting-started/accepting-new-projects/#accepting-new-projects

hathach, 2022-12-06:
> Sweet, I'll open a WIP progress PR over at OSS-fuzz to integrate tinyusb. Is your '@tinyusb.org` email address associated with a google account? I'm just going over the requirements for accepting [...]

unfortunately not, it is an office 365 email provided by hosting company. I have just created an new gmail for this purpose at `tinyusb.org@gmail.com` , I will try to make a few commit with this new email if that make verification steps easier. Thank you.

silvergasp, 2022-12-06:
Easy I've added a draft initial integration here https://github.com/google/oss-fuzz/pull/9143 but I'll wait until the harnesses are merged here otherwise there isn't much point.

silvergasp, 2022-12-08:
I'm going to work on an HID device fuzzer next up. I'll keep you posted here as to what I'm working on so that we don't overlap.

silvergasp, 2022-12-08:
Also I'll see if I can get the oss-fuzz integration working shortly too!

hathach, 2022-12-11:
@silvergasp we probably need to update the oss-fuz pr to match the new folder location of fuzz in `test/fuzz`

silvergasp, 2022-12-12:
> @silvergasp we probably need to update the oss-fuzz pr to match the new folder location of fuzz in `test/fuzz`

Yeah I'll take a better look tonight

silvergasp, 2022-12-12:
Just to confirm have you made any commits with that gmail account?

silvergasp, 2022-12-12:
I've marked the OSS-fuzz integration as ready to merge https://github.com/google/oss-fuzz/pull/9143. Hopefully everything is in order and it get's merged relatively soon.

hathach, 2022-12-12:
> Just to confirm have you made any commits with that gmail account?

yeah, I did, just made all recent merged and commits with that gmail account.

silvergasp, 2022-12-28:
So I've got some CI feedback from the OSS-fuzz repo, and it looks like there was a few build failures. I've put up a PR here that should mostly address these problems.

https://github.com/hathach/tinyusb/pull/1823

hathach, 2023-01-13:
@silvergasp thank you very much, tinyusb is accepted to oss-fuzz. Though strangely, I would expect an notification email from oss-fuzz but received none so far. Also the document said we could check build status after 1 day or so, but couldn't find it https://oss-fuzz-build-logs.storage.googleapis.com/index.html . 

There is no rush, I just wonder if there is anything we should do or just simply wait for a few more days for it to get run.

silvergasp, 2023-01-13:
Yeah, I actually think this is an upstream bug in oss fuzz. It turns out that many new projects are not working properly. There is a similar tracking issue here;

https://github.com/google/oss-fuzz/issues/9408

While the name of the issue is not descriptive of tinyusb it appears to be an identical problem.

hathach, 2023-01-18:
thank you @silvergasp , seems like they fix the issue already. tinyusb build log can be found at https://oss-fuzz-build-logs.storage.googleapis.com/index.html#tinyusb

silvergasp, 2023-01-18:
Yeah, it's looking like things are on track again. I don't think the fuzzer's have been run yet, but I'll check again tomorrow to see what's going on. We should be able to see some statistics once it's been run here;

https://oss-fuzz.com/fuzzer-stats?project=tinyusb&fuzzer=afl&job=afl_asan_tinyusb&group_by=by-fuzzer

You'll need to log in with your tinyusb.org@gmail.com email account to get access to that dashboard.

silvergasp, 2023-01-19:
Looks like it's running well. I've got a little bit of tweaking to get it working well with the AFL fuzzing engine. But other than that I'm super impressed that it hasn't found any new bugs. Its already done more than 1000x the iterations than i ever did on my laptop. Their servers are a lot quicker. You've done well writing tinyusb to be so robust against fuzzing!

hathach, 2023-01-19:
ah thanks, the dashboard starts to show some interesting statistic, though to be honest, I don't know these well enough.  

> Looks like it's running well. I've got a little bit of tweaking to get it working well with the AFL fuzzing engine. But other than that I'm super impressed that it hasn't found any new bugs. Its [...]

I hope so, but there is chance that we didn't have enough fuzzing. I will try to learn more of fuzzing to add more testing in the future. Thank you again, this will hugely improve the stability of tinyusb.

silvergasp, 2023-01-19:
Not a problem :)

Well here's a more verbose rundown. There are multiple sanitizers (you can think of these as bug detectors). The ones that we have enabled with OSS-fuzz are;
- Address sanitizer (ASAN), which among other things will detect buffer overflows.
- Memory sanitizer (MSAN), which among other things will detect usage of uninitialised memory.
- Undefined behaviour sanitizer (UBSAN), which will detect undefined behaviour e.g. an integer overflow.

Then we've got multiple fuzzing engines enabled. Each fuzzing engine behaves a little differently and will mutate the inputs differently. The overall concept is the same it's just how the fuzzing is performed that is different. We have a few of these enabled;
- [American fuzz lop (AFL)](https://lcamtuf.coredump.cx/afl/)
- [Honggfuzz](https://honggfuzz.dev/)
- [LLVM libfuzzer](https://www.llvm.org/docs/LibFuzzer.html)

So when you look at the dashboard you'll see a kind of matrix expansion of sanitizers x fuzzing engine.

For the most part, these are all working great for the mass storage, cdc, and net classes. However, if we look at the stats for the AFL fuzzer you'll see that the ["statibility" is pretty low](https://oss-fuzz.com/fuzzer-stats?group_by=by-day&fuzzer=afl_tinyusb_cdc&job=afl_asan_tinyusb&project=tinyusb), (27.9 out of 100). This isn't fantastic and it should really be closer to 90/100, to get good results from AFL. The reason why stability is so low is that AFL makes the assumption that it can run, multiple iterations in parallel threads. However, in the case of tinyusb this is a poor assumption, as there is a fairly significant amount of global state (variables). This means that there are instead a bunch of different threads all modifying the same global state rather than having their own independent instance of tinyusb. I have a bit of an idea of how we can address this e.g. something like this;
```C
#ifdef FUZZ
   // Make a thread-local static global so that each thread has it's only local instance of tinyusb
   #define TU_STATIC __thread static
#else 
   #define TU_STATIC static
#endif
```

It'll then just be a process of going through all the `static` vars and replacing them with `TU_STATIC`. I'm part way through this process, so I'll have a PR up soon.

hathach, 2023-01-19:
thank you very much for the very detail explanation. Things are much clearer to me now. TU_STATIC look good, we can then wrap all variable into a struct to make it easier to add TU_STATIC. I definitely need to learn more about fuzzing in the future.

silvergasp, 2023-01-20:
Yeah there are two options;
- We either centralise all of the mutable static globals and make that thread local
- We make all mutable static globals __thread static so that they are thread local

silvergasp, 2023-01-20:
On a different note, the code coverage for the msc and net class fuzzers is not fantastic and i might be missing something in my fuzzing stubs. I'm going to take a closer look over the next couple of days, but if you've got time to take a quick look at the fuzz harnesses/callbacks that would be fantastic.

https://storage.googleapis.com/oss-fuzz-coverage/tinyusb/reports/20230119/linux/src/tinyusb/src/class/report.html

hathach, 2026-11-06:
Fixed: fuzz harnesses landed in aedae6201, 92b5f2d2e and 6492f4a18 (now `test/fuzz`, built by OSS-Fuzz). The remaining gaps are tracked in #4220-#4223.

closed 2026-11-06 by hathach