hathach/tinyusb #1721: Double-buffered bulk transfers in RP2040 USB Host Mode do not complete

rppicomidi, 2022-11-05:
### Operating System

Others

### Board

Raspberry Pi Pico

### Firmware

I am working on a mass storage class demo program [here](https://github.com/rppicomidi/pico-usb-host-msc-demo). I am able to mount a flash drive, list directories and display file contents, but only if double-buffering is disabled.

### What happened ?

When the fatfs code attempted to read the 512 byte boot sector from the flash drive, the code hung up waiting for the transfer
callback to be called.

### How to reproduce ?

1. Build the code as described [here](https://github.com/rppicomidi/pico-usb-host-msc-demo) except do not apply the patch to tinyusb
2. Hook up the hardware as described.
3. Load the code image to the Pico board.
4. Start the program and observe the prompts on the terminal
5. Set the date and time as required
6. Attach a USB flash drive.
7. Observe it mounted correctly
8. Type the `ls` command
9. Observe the code hangs up.

### Debug Log as txt file

I did not generate a debug log. I discovered where the code hung up using an interactive debugger.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

rppicomidi, 2022-11-05:
@hathach I can submit my patch as a pull request if you would like.

hathach, 2022-11-17:
sorry for late response as usual, yeah, I remembered seeing some issue with double buffering in device (out endpoint), and thought that would also apply to host in as well 

https://github.com/hathach/tinyusb/blob/master/src/portable/raspberrypi/rp2040/rp2040_usb.c#L155-L157

I forgot the detail, but disabling double buffered is not really an option here due to rp2040 Errata E4 #891 (also it hurts throughput). The optimal solution is figuring out what is wrong with double buffered and fix it accordingly. Btw, which tinyusb patch you are mentioning ?

rppicomidi, 2022-11-17:
As I recall Errata E4 has a workaround for single-buffered transactions, which you have already implemented in `sync_ep_buffer()`. I agree disabling double-buffering hurts throughput, but from my testing, it does not seem to hurt functionality.

The current code in `_hw_endpoint_start_next_buffer()` is

```
  // For now: skip double buffered for Device mode, OUT endpoint since
  // host could send < 64 bytes and cause short packet on buffer0
  // NOTE this could happen to Host mode IN endpoint
  bool const force_single = !(usb_hw->main_ctrl & USB_MAIN_CTRL_HOST_NDEVICE_BITS) && !tu_edpt_dir(ep->ep_addr);
```

Will you point out where you got the information that Errata E4 only applies to OUT endpoints in device mode and IN endpoints in host mode? If that is correct, I can make a better patch. However, I am not seeing it in the specification's description of E4:

```
The USB host maintains a buffer selector which switches between BUF0 and BUF1. This should only be
toggled in double buffered mode but is toggled in single buffered mode too. For a transaction lasting
multiple packets (i.e. length more than 8 bytes in low speed mode, and length more than 64 bytes in full
speed mode), the buffer status can be written back to the BUF1 half of the status register when the buffer
select is incorrectly set to BUF1. Note this does not affect reading new buffer information from the buffer
control register, as the controller ignores the buffer selector in single buffered mode when reading the
buffer control register.
```

rppicomidi, 2022-11-18:
@hathach 
I just tried the following code patch and it worked for MSC. If I only disabled double-buffering for IN endpoints, then
IN endpoints work fine, but OUT transactions do not complete; f_mkdir(), for example, locks up.

```
  // For now: skip double buffered for Device mode, OUT endpoint since
  // host could send < 64 bytes and cause short packet on buffer0
  // NOTE this could happen to Host mode IN endpoint
  bool const force_single = (!(usb_hw->main_ctrl & USB_MAIN_CTRL_HOST_NDEVICE_BITS) && !tu_edpt_dir(ep->ep_addr)) ||  // device OUT endpoint
    ((usb_hw->main_ctrl & USB_MAIN_CTRL_HOST_NDEVICE_BITS)); // all host endpoints
```

hathach, 2022-11-18:
E4 requires a walkaround when using host with single buffered mode, which is removed when double buffered is implemented. From my experience, the walkaround has its own bizarre issue and not working reliably as well. As mentioned above, going back to single buffered + E4 + walkaround is not a good approach. The optimal solution is fixing what is breaking with double buffered, maybe we just miss some logic/buffer/sync/schedule etc ... 

btw, I really like your msc demo, I have written an simpler msc navigation but it is not that good. I will try to add an official msc example based on your demo to make it easier to reproduce and fix this issue in the future.

hathach, 2022-11-21:
I just added an host msc file explorer example which work well with pio-usb but indeed has issue with native usb. The new example will probably help with troubleshoot this issue later on. Unfortunately, I have to switch to other work and will come back to this later on.

rppicomidi, 2022-11-22:
OK. Something to note when you get back to it. From the RP2040 datasheet, double buffer does not work for Host Interrupt
endpoints. I presume the same applies for Host bulk endpoints given how they are implemented now.

4.1.2.7.4. Interrupt Endpoints
The host controller can poll interrupt endpoints on many devices (up to a maximum of 15 endpoints). To enable these,
the programmer must:
• Pick the next free interrupt endpoint slot on the host controller (starting at 1, to a maximum of 15)
• Program the appropriate endpoint control register and buffer control register like you would with a normal IN or OUT
transfer. **Note that interrupt endpoints are only single buffered so the BUF1 part of the buffer control register is
invalid.**

hathach, 2022-11-23:
@rppicomidi yeah, I think you probably hit the nail on the head. I totally forget about this, and didn't pay attention to this. In this case we should disable double buffering for non-control endpoint with the current implementation + E4 walkaround. Hopefully this could buy us time to better support bulk endpoint without using interrupt endpoint for the sake of throughput later on.

rppicomidi, 2022-11-23:
I have submitted pull-request #1755 to fix this issue.

hathach, 2022-11-30:
implemented by #1755 , thank you very much

closed 2022-11-30 by hathach