hathach/tinyusb #1722: audio 4 channel mic example transmits only 2 channels

clwe, 2022-11-05:
### Operating System

MacOS

### Board

raspberrypi pico rp2040

### Firmware

examples/device/audio_4_channel_mic

### What happened ?

The device succesfully shows up as a 4 channel USB device, but only 2 of 4 channels transmit the test data. 

- Channel 1 & 3 transmit a low amplitude saw tooth wave (as expected)
- Channel 2 & 5 transmit total silence

### How to reproduce ?

compile & flash example to rp2040
open in DAW (tested in Audacity & Ableton Live)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

jh2srv, 2023-01-22:
Hello, 
I can confirm this bug occurs also in Windows 10 and the RP2040 (Raspberry pi pico)

PanRe, 2023-02-16:
Most likely, this is a problem within the RP2040 device controller driver (DCD), where the handling of the ISO EP may not be correct. This was at least the problem for other MCUs. The UAC driver is most likely not to blame, as it works for many MCUs. I did not investigate the RP2040 DCD however. The DCD is furthmore subject to improvements e.g. see #540. If you have time, please feel free to solve this issue and make PR! Thx in advance :)

nrg83, 2025-01-24:
Hello, I also observe this behavior on a STM32F411CEU6 ([WeActStudio.MiniSTM32F4x1](https://github.com/WeActStudio/WeActStudio.MiniSTM32F4x1)). I am on release 0.18.0 and use examples\device\audio_4_channel_mic. In Audacity I only see CH0 (saw wave) and CH1 (inverted saw wave) with small amplitude. I defined CFG_TUSB_MCU as OPT_MCU_STM32F4. Is there anything else I need to consider?
My OS is Windows 10.

If you have any suggestions on what I could try, I'm happy to test it on my hardware if you don't have it available.