hathach/tinyusb #1728: UAC2 Headset example fails after some playback time

angmolin, 2022-11-07:
### Operating System

Others

### Board

RP2040

### Firmware

examples/device/uac2_headset

### What happened ?

After upload the `.uf2` file from `uac2_headset` example compilation it works, but after changing a song or keep listening for a while it fails calling the function `tud_audio_set_itf_cb` with parameters `interface 1 alt X` (alt varies between 0~2) but it always fails on this function. I also connected an UART to the pins 0 and 1 of the raspberry and it says `*** PANIC *** Hard assert` I will try to find the cause of the problem and provide a solution for it.

### How to reproduce ?

1. Compile the `uac2_headset` example for the Raspberry Pi Pico (not H or W in my case)
2. Copy the `.uf2` file
3. Select TinyUSB headset as output on configuration
4. Open your favorite music player and wait until playback freezes

### Debug Log as txt file

tud_mount_cb:50 : USB device mounted!
tud_audio_clock_get_request:87 : Clock get 2 freq ranges
tud_audio_clock_get_request:101 : Range 0 (44100, 44100, 0)
tud_audio_clock_get_request:101 : Range 1 (48000, 48000, 0)
tud_audio_clock_get_request:87 : Clock get 2 freq ranges
tud_audio_clock_get_request:101 : Range 0 (44100, 44100, 0)
tud_audio_clock_get_request:101 : Range 1 (48000, 48000, 0)
tud_audio_set_itf_cb:317 : Set interface 1 alt 1
tud_audio_set_itf_cb:321 : Audio streaming started!
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_clock_get_request:75 : Clock get current freq 44100
tud_audio_clock_set_request:138 : Clock set current freq: 48000
tud_audio_clock_get_request:75 : Clock get current freq 48000
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0
tud_audio_set_itf_cb:317 : Set interface 1 alt 1
tud_audio_set_itf_cb:321 : Audio streaming started!
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0
tud_audio_clock_get_request:87 : Clock get 2 freq ranges
tud_audio_clock_get_request:101 : Range 0 (44100, 44100, 0)
tud_audio_clock_get_request:101 : Range 1 (48000, 48000, 0)
tud_audio_clock_get_request:87 : Clock get 2 freq ranges
tud_audio_clock_get_request:101 : Range 0 (44100, 44100, 0)
tud_audio_clock_get_request:101 : Range 1 (48000, 48000, 0)
tud_audio_set_itf_cb:317 : Set interface 2 alt 1
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_clock_get_request:75 : Clock get current freq 48000
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_set_itf_cb:317 : Set interface 2 alt 0
tud_audio_feature_unit_get_request:182 : Get channel 1 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 1 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 1 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 1 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:189 : Get channel 1 volume 0 dB
tud_audio_feature_unit_set_request:227 : Set channel 1 volume: -1 dB
tud_audio_feature_unit_get_request:189 : Get channel 1 volume -1 dB
tud_audio_feature_unit_set_request:227 : Set channel 1 volume: 0 dB
tud_audio_feature_unit_get_request:182 : Get channel 0 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 0 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 0 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:182 : Get channel 0 volume range (-50, 0, 1) dB
tud_audio_feature_unit_get_request:189 : Get channel 0 volume 0 dB
tud_audio_feature_unit_set_request:227 : Set channel 0 volume: -1 dB
[... 129 lines cut ...]
tud_audio_clock_get_request:75 : Clock get current freq 48000
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0
tud_audio_set_itf_cb:317 : Set interface 1 alt 1
tud_audio_set_itf_cb:321 : Audio streaming started!
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0
tud_audio_set_itf_cb:317 : Set interface 1 alt 1
tud_audio_set_itf_cb:321 : Audio streaming started!
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_clock_get_request:75 : Clock get current freq 48000
tud_audio_clock_set_request:138 : Clock set current freq: 44100
tud_audio_clock_get_request:75 : Clock get current freq 44100
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0
tud_audio_set_itf_cb:317 : Set interface 1 alt 1
tud_audio_set_itf_cb:321 : Audio streaming started!
tud_audio_clock_get_request:110 : Clock get is valid 1
tud_audio_set_itf_close_EP_cb:305 : Audio streaming stopped!
tud_audio_set_itf_cb:317 : Set interface 1 alt 0

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

holysnippet, 2024-02-14:
Hello, I think I have the same problem. Have you found any leads since then?

kholia, 2024-04-12:
I believe this is fixed by https://github.com/hathach/tinyusb/pull/1802. Give it a go, and do report back.

HiFiPhile, 2024-04-12:
A new ISO endpoints allocation API was introduced to resolve this issue, it's implemented in audio and video class and dcd driver of stm32, esp32 etc. But nobody works on rp2040 driver yet.

closed 2026-11-14 by HiFiPhile