hathach/tinyusb #1729: Vulnerability reporting procedure

silvergasp, 2022-11-07:
### Related area

Security/vulnerability reporting

### Hardware specification

N/A

### Is your feature request related to a problem?

I think I've found a security vulnerability, but I don't want to publicly describe the problem on a GitHub issue until the problem is fixed. Having a security problem published publicly would open people up to attack.

### Describe the solution you'd like

I'd like some procedures to be in place to;
- Contact the maintainers of this repository with security issues without publicly posting the problem to a GitHub issue. e.g. https://github.com/hathach/tinyusb/security/policy
- Post security problems publicly once they have been fixed, and in the latest release. e.g. https://github.com/hathach/tinyusb/security/advisories

NOTE: There are some templates on the links above which offer a good starting point.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

cr1901, 2022-11-15:
I agree something should be done about this, apparently Github now has a feature to contact project maintainers securely/directly in case of a vuln? Idk much about it tho...

silvergasp, 2022-11-15:
I wasn't aware of that. But it looks like a great option!

Link to the docs for posterity;

https://docs.github.com/en/code-security/security-advisories/repository-security-advisories/configuring-private-vulnerability-reporting-for-a-repository

silvergasp, 2022-11-15:
It appears pretty easy to setup. But it would need to be done by someone with admin privileges.

cr1901, 2022-11-15:
Well, @hathach seems to be taking a break for now, but I imagine he'll be back soon. Thanks for the link, since I completely forgot other than hearing about the new feature thirdhand :D!

silvergasp, 2022-11-15:
Yeah, I've emailed @hathach directly with directly with details of the vulnerability that I found. Got a reply mentioning that he was keen to take a look, but is currently away from his computer. 

I don't think there is any need to rush into getting it fixed :). Though it would be nice to get a fix together in the next couple of weeks.

hathach, 2022-11-16:
just be back to my PC, will check this out in this week or so. Thank you for the issue/pr.

hathach, 2023-01-07:
this should be fixed by #1789, thank you @silvergasp very much for bringing this up and also making effort to add fuzzer as well and pr to integrate tinyusb to google fuzzing.

closed 2023-01-07 by hathach