hathach/tinyusb #1730: RP2040-Zero fails to show up as a USB device after disconnect

jthlim, 2022-11-07:
### Operating System

Others

### Board

RP2040-Zero

### Firmware

examples/device/hid_boot_interface

But really, every tinyusb example I tested exhibited the same problem.

### What happened ?

On RP2040-Zero, immediately after flashing the firmware, a TinyUSB device shows up and can interact with the computer.

However, upon disconnecting and reconnecting, the device no longer shows. Plugging it in to other devices also do not recognize the TinyUSB device.

Have noticed this with both RP2040-Zeros I own, and have seen this issue also on a Uni v4 -- a rp2040 based Steno Keyboard.

Not reproducible on RPi Pico or WeAct Studio Pico boards -- disconnecting and reconnecting, or plugging them into other devices works as expected.

### How to reproduce ?

1. Build example/device/hid_boot_interface using github.com/raspberrypi/pico-examples
2. copy firmware to device
3. Observe device is working as expected
4. Unplug
5. Plug back in
6. Notice that device is not working, and that the device does not show up in system information/

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

jthlim, 2022-11-09:
@kilograham Tagging Graham as it seems he may have written some of this code.

Note: Bootloader shows reliably the USB device, and the original QMK firmware on the Steno keyboard works fine; just the tiny usb implementation only works after a reboot.

jthlim, 2022-11-09:
Update: The rp2040 zero issues seem to be separate; the issue with tiny_usb devices not showing on the steno board persist though; and are intermittent.. most times it doesn't work; but when connecting to a sleeping machine, it seems to be able to wake it up *most* of the time, then work correctly after.

jthlim, 2022-11-10:
Going to close this; ended up being a flash spi clkdiv issue.

closed 2022-11-10 by jthlim

AlynxZhou, 2023-06-11:
> Update: The rp2040 zero issues seem to be separate (seems like they don't boot fullstop after disconnecting); the issue with tiny_usb devices not showing on the steno board persist though; and are intermittent.. most times it doesn't work; but when connecting to a sleeping machine, it seems to be able to wake it up _most_ of the time, then work correctly after.

I am sorry to bother you but I got the same problem with you, I have 3 RP2040-Zero from WaveShare, only 1 works, the other 2 cannot be detected by host after reconnecting, do you have workarounds for it? Thanks.

AlynxZhou, 2023-06-11:
> Going to close this; ended up being a flash spi clkdiv issue.

OK, I find the reason and submit a fix to the pico-sdk: <https://github.com/raspberrypi/pico-sdk/pull/1421>.