hathach/tinyusb #1732: PIO USB Host does not wait T_RSTRCY before starting enumeration

rppicomidi, 2022-11-09:
### Operating System

Others

### Board

Raspberry Pi Pico

### Firmware

I discovered this whilst experimenting with MSC host. I have my own [demo code](https://github.com/rppicomidi/pico-usb-host-msc-demo/tree/main/PIO-USB-Demo).

### What happened ?

See issue described [here](https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/52). Code is built with Log level is 1, when
you plug in a USB drive, the control transfer fails and the terminal output is
```
[1:0] Control FAILED
```
If Log level is 3, there is a lot of debug output but the flash drive successfully mounts.

I can attach screenshots of the logic analyzer traces if it helps. However, what I measured is less than 1ms between the end of bus reset to the start of the first setup packet when log level is 1. When the log level is 3, I measure about 8.8ms between reset end and the first setup packet. When I read USB 2.0 spec section 7.1.7.3 Figure 7-29 timing interval delta-t6 must be `T_RSTRCY` which is 10ms. 

I can fix the issue by adding a 10ms delay to `hcd_port_reset_end()` as follows:

```
void hcd_port_reset_end(uint8_t rhport)
 {
   uint8_t const pio_rhport = RHPORT_PIO(rhport);
   pio_usb_host_port_reset_end(pio_rhport);
+  osal_task_delay(10); // 10ms reset recovery time (USB 2.0 spec 7.1.7.4 delta-t6)
 }
```

Would you like me to submit a pull request?

### How to reproduce ?

Build the [demo code](https://github.com/rppicomidi/pico-usb-host-msc-demo/tree/main/PIO-USB-Demo) without first applying
the patch to the tinyusb code.

Run the code on the Pico board.

Plug in a USB flash drive

Observe

```
[1:0] Control FAILED
```
unless the log level is 3.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-11-21:
The delay is already done by usbh.c 
https://github.com/hathach/tinyusb/blob/master/src/host/usbh.c#L1263

I have no issue with mounting msc with pio-usb. Could you give the new example a try #1746 to see if that work for you. Note: you need to change CFG_TUH_RPI_PIO_USB =1 in tusb_config.h

hathach, 2022-11-28:
I just encounter this issue while working on other task. Could you confirm the failed control occurred with the very first control by adding TU_LOG_LOCATION() here (with LOG=1)

https://github.com/hathach/tinyusb/blob/081e4eecba3c91f08cdf96b0a090e7726cec3727/src/host/usbh.c#L1187

If yes, then the issue is not host stack not waiting, since stack already wait 500ms (longer than USB specs requires). The bug is caused by device side start up is taking too long. A proper fix would be retry a few times as suggested by the comment.

PR #1763 (2022-11-28): Retry a few times with transfers in enumeration since device can be unstable when starting up
**Describe the PR**
Some device has large starting up and slow to response to first few request. This PR retry a few times with transfers in enumeration. Should fix #1732
closed 2022-11-28 by hathach