hathach/tinyusb #1741: rp2040: ep 0 out was already available

DatanoiseTV, 2022-11-16:
### Operating System

Linux

### Board

Custom RP2040 board

### Firmware

https://github.com/rppicomidi/midi2usbhost custom version

### What happened ?

I've built a custom board which has USB Type-C for Firmware Uploads (and during boot) and a USB Type-A which is switched to the D+/D- Pins with a FSUSB42 (and which power is controlled via MOSFET).

The select pin of the FSUSB42 is pulled low, so the device starts up with the USB-C being active for firmware upload.
Then the firmware puts SEL high, calls board_init() / tusb_init() as usual and enables VBUS. The USB-C is then only used for supplying the board with power.

Unfortunately and unexpectedly, the Type-A won't work because TinyUSB says the EP 0 out was already claimed.

I've tried a workaround with calling a usb reset before calling tusb, but that won't help.
```cpp
    // set device to USB Host mode
    gpio_put(PIN_USB_MUX_SEL, 1);

        // Reset usb controller
    reset_block(RESETS_RESET_USBCTRL_BITS);
    unreset_block_wait(RESETS_RESET_USBCTRL_BITS);

    // Clear any previous state in dpram just in case
    memset(usb_dpram, 0, sizeof(*usb_dpram)); // <1>

    gpio_init(PIN_VUSB_EN);
    gpio_set_dir(PIN_VUSB_EN, GPIO_OUT);
    gpio_put(PIN_VUSB, 1);
```

### How to reproduce ?

1. Clone the midi2usbhost repo
2. Try to switch from a device being connected on board to a host being connected

### Debug Log as txt file

ep 0 out was already available

Configured MIDI UART 1 for 31250 baud
WARN: starting new transfer on alreaassertion "!(buf_ctrl & USB_BUF_CTRL_FULL)" failed: file "/home/syso/dev/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c", line 219, function: sync_ep_buffer

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

DatanoiseTV, 2022-11-17:
On further investigation, this seems to happen with some MIDI devices and unrelated to my USB MUX. For example, I cannot get a Novation Remote Zero SL to enumerate.

hathach, 2022-11-30:
move to discussion since it is not clear too me if the bug is caused by tinyusb stack or user error