hathach/tinyusb #1747: Port for Windows UDE?

203Null, 2022-11-22:
### Related area

New Port Support

### Hardware specification

Windows 10 PC

### Is your feature request related to a problem?

So projects using TinyUSB can be tested on a Windows PC without hardwares

### Describe the solution you'd like

https://github.com/microsoft/UDE 

Seems like Windows 10 supports USB device emulation, does this mean TinyUSB can be ported on top of it and presented to Windows as a virtual USB device?

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-11-23:
I am new (clueless) to this microsoft UDE. But it sounds like a great way to emulate TinyUSB with windows. Also have you tried running Microsoft device examples, it would be helpful to detail the step to run/test since I am mostly on Linux and only has an decade-old windows laptop.

203Null, 2022-11-23:
> I am new (clueless) to this microsoft UDE. But it sounds like a great way to emulate TinyUSB with windows. Also have you tried running Microsoft device examples, it would be helpful to detail the step to run/test since I am mostly on Linux and only has an decade-old windows laptop.   

No yet. I wish I know more about how USB works at a lower level to port it myself.