hathach/tinyusb #1798: OSAL FreeRTOS queue usage lead to wrong behavior of tasks

jbruneaux31, 2022-12-15:
### Operating System

Others

### Board

Custom ARM CM33 board

### Firmware

TinyUSB configured for one CDC and one USB Audio link

### What happened ?

Lets describe a simple setup with 2 FreeRTOS tasks (Task1= TinyUSB, Task 2 = other process), with FreeRTOS scheduler not using time slicing (i.e same priorities tasks will run only when the other task will either suspend or yield):
- Task2 is running
- USB interrupt happens for audio EPOUT, which enqueue data using osal_queue_send function.
- Task2 should continue to run until it voluntarily yield or it suspends

We observe that this scenario is not always true. Most of the time, after the interrupt happens, we see a context switch and Task1 runs, which is not what is expected.

Investigation has shown that in functions osal_queue_send & osal_semaphore_post, the xHigherPriorityTaskWoken variable in not initialised (use the stack, so unpredictable content). This variable is passed to some FreeRTOS functions which will only set this variable to pdTRUE if task switch is required, but will not touch the variable otherwise.

The fix is to set xHigherPriorityTaskWoken to pdFALSE in those two functions.

### How to reproduce ?

See description above. The best way to observe the problem is to do have an activable endless loop in Task2 so that after USB initialisation, the problem can be observed

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-12-16:
it looks like a bug, since you are at it and have your setup testing it. Would you mind making an PR to fix this bug ?

jbruneaux31, 2022-12-16:
Well, sorry I'm pretty new to this and will try to find how to make this PR. Will probably be able to do that in the next week.

jbruneaux31, 2022-12-19:
Hi,

I tried to make a pull-request. Let me know if this worked.

Regards,

hathach, 2022-12-20:
Fixed by #1803

closed 2022-12-20 by hathach