hathach/tinyusb #1808: N-Key-Rollover

LuggaFiech, 2022-12-21:
### Related area

No NKRO possible with tinyusb?

### Hardware specification

rp2040 feather

### Is your feature request related to a problem?

I‘m currently facing the issue that I am unable to get N key rollover to work. If there is any way to do it currently please tell me.

### Describe the solution you'd like

I don‘t know any solution.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

dhalbert, 2022-12-21:
NKRO support really has to do with your HID report descriptor, and is not a TinyUSB issue. There's an example of NKRO in CircuitPython here: https://learn.adafruit.com/custom-hid-devices-in-circuitpython/n-key-rollover-nkro-hid-device. CircuitPython uses TinyUSB internally.

RobertDaleSmith, 2023-09-01:
What about in the case of using TinyUSB as a host device reading a HID keyboard? Appears the built-in support is for hid keyboard boot mode which reports only 6 keys. From what I understand there is a command that must be sent to a keyboard from a host device to enable full NKRO mode. Is this correct and are there any known examples?