hathach/tinyusb #1811: tud_cdc_n_peek/tu_fifo_peek don't have an offset parameter as documentation says

pho, 2022-12-23:
### Operating System

Linux

### Board

Any

### Firmware

This happens on current TinyUSB master branch

### What happened ?

Both of these functions imply in their docstring that you can peek at any specific position in the FIFO, but then the offset parameter is missing

```
// Get a byte from FIFO at the specified position without removing it
bool     tud_cdc_n_peek            (uint8_t itf, uint8_t* ui8);
```
https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.h#L83

```
/******************************************************************************/
/*!
    @brief Read one item without removing it from the FIFO.
    This function checks for an overflow and corrects read pointer if required.
    @param[in]  f
                Pointer to the FIFO buffer to manipulate
    @param[in]  offset
                Position to read from in the FIFO buffer with respect to read pointer
    @param[in]  p_buffer
                Pointer to the place holder for data read from the buffer
    @returns TRUE if the queue is not empty
 */
/******************************************************************************/
bool tu_fifo_peek(tu_fifo_t* f, void * p_buffer)
```
https://github.com/hathach/tinyusb/blob/master/src/common/tusb_fifo.c#L678

### How to reproduce ?

Not needed

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2022-12-23:
thanks for reporting, index is used to be supported by removed recently due to lack of usage. I will update the doc later on.

pho, 2022-12-23:
Oh, thank you very much for your quick response

closed 2022-12-23 by pho