hathach/tinyusb #1830: tuh_hid_set_report fails after multple calls to change a keyboards LEDs. Number of calls is non determinsitic

morio, 2022-12-31:
### Operating System

Others

### Board

Custom RP2040

### Firmware

https://github.com/rabbitholecomputing/RP2040-TinyUsb-keyboard-leds

### What happened ?

After multiple calls toggling a USB keyboard LEDs `tuh_hid_set_report` starts returning false.

Tracing my original code (https://github.com/rabbitholecomputing/QuokkADB-firmware) the issue is [_ctrl_xfer.stage == CONTROL_STAGE_IDLE);](https://github.com/hathach/tinyusb/blob/7004914d8c6c7d88908776eb153661b6b987e513/src/host/usbh.c#L518)
fails because _ctrl_xfer.stage is equal to CONTROL_STAGE_ACK.

 The test code doesn't actually exercise that line because it uses `tuh_hid_set_report_complete_cb` to initiate another   `tuh_hid_set_report`.  And since tuh_hid_set_report never finishes it can't be called again in the test code.

### How to reproduce ?

1. Upload firmware to RP2040 TinyUSB host connected to a USB keyboard
2. Hook up board to a pico probe
3. Open pico probe COM port in a terminal like Tera Term
4. Power board
5. Press "Caps Lock"
6. Watch the output fly by in a terminal window
7. When the LEDs stop blinking tuh_hid_set_report has failed.
8. The terminal should output "TEST: Set report completed, Toggled leds X times so far." Where X is the number of successful LED toggles.

### Debug Log as txt file

The last three toggles of the LEDs using `tuh_hid_set_report`. The very last one hangs and never completes.
Full in the .log file linked at the bottom

```
HID Set Report: id = 0, type = 2, len = 1

[0:1] Class Request: 21 09 00 02 00 00 01 00

hw_endpoint_init dev 1 ep 0 out xfer 0

dev 1 ep 0 out setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

Transfer complete

Sent setup packet

on EP 00 with 8 bytes

hcd_edpt_xfer dev_addr 1, ep_addr 0x0, len 1

  Prepare BufCtrl: [0] = 0xf401  [1] = 0x0000

Buffer complete

buf_status 0x00000001

Single Buffered: ep_ctrl = A0000180

  Sync BufCtrl: [0] = 0x6001  [1] = 0x0000

  Short packet on buffer 0 with 1 bytes

Completed transfer of 1 bytes on ep 0 out

Transfer complete

on EP 00 with 1 bytes

[0:1] Control data:

  0000:  07                                               |.|

hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 0

hw_endpoint_init dev 1 ep 0 in xfer 0

dev 1 ep 0 in setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000

Buffer complete

buf_status 0x00000001

Single Buffered: ep_ctrl = A0000180

  Sync BufCtrl: [0] = 0xe000  [1] = 0x0000

  Short packet on buffer 0 with 0 bytes

Completed transfer of 0 bytes on ep 0 in

Transfer complete

on EP 80 with 0 bytes

HID Set Report complete

TEST: Set report completed, Toggled leds 211 times so far.

HID Set Report: id = 0, type = 2, len = 1

[0:1] Class Request: 21 09 00 02 00 00 01 00

hw_endpoint_init dev 1 ep 0 out xfer 0

dev 1 ep 0 out setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

Transfer complete

Sent setup packet

on EP 00 with 8 bytes

hcd_edpt_xfer dev_addr 1, ep_addr 0x0, len 1

  Prepare BufCtrl: [0] = 0xf401  [1] = 0x0000

Buffer complete

buf_status 0x00000001

Single Buffered: ep_ctrl = A0000180

  Sync BufCtrl: [0] = 0x6001  [1] = 0x0000

  Short packet on buffer 0 with 1 bytes

Completed transfer of 1 bytes on ep 0 out

Transfer complete

on EP 00 with 1 bytes

[0:1] Control data:

  0000:  00                                               |.|

hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 0

hw_endpoint_init dev 1 ep 0 in xfer 0

dev 1 ep 0 in setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000

Buffer complete

buf_status 0x00000001

Single Buffered: ep_ctrl = A0000180

  Sync BufCtrl: [0] = 0xe000  [1] = 0x0000

  Short packet on buffer 0 with 0 bytes

Completed transfer of 0 bytes on ep 0 in

Transfer complete

on EP 80 with 0 bytes

HID Set Report complete

TEST: Set report completed, Toggled leds 212 times so far.

HID Set Report: id = 0, type = 2, len = 1

[0:1] Class Request: 21 09 00 02 00 00 01 00

hw_endpoint_init dev 1 ep 0 out xfer 0

dev 1 ep 0 out setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

Transfer complete

Sent setup packet

on EP 00 with 8 bytes

hcd_edpt_xfer dev_addr 1, ep_addr 0x0, len 1

  Prepare BufCtrl: [0] = 0xf401  [1] = 0x0000

Buffer complete

buf_status 0x00000001

Single Buffered: ep_ctrl = A0000180

  Sync BufCtrl: [0] = 0x6001  [1] = 0x0000

  Short packet on buffer 0 with 1 bytes

Completed transfer of 1 bytes on ep 0 out

Transfer complete

on EP 00 with 1 bytes

[0:1] Control data:

  0000:  07                                               |.|

hcd_edpt_xfer dev_addr 1, ep_addr 0x80, len 0

hw_endpoint_init dev 1 ep 0 in xfer 0

dev 1 ep 0 in setup buffer @ 0x0x50100180

endpoint control (0x0x50100100) <- 0xa0000180

  Prepare BufCtrl: [0] = 0x7400  [1] = 0x0000
```

[set_report_fails.log](https://github.com/hathach/tinyusb/files/10326639/set_report_fails.log)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-03-22:
> Tracing my original code (https://github.com/rabbitholecomputing/QuokkADB-firmware) the issue is [_ctrl_xfer.stage == CONTROL_STAGE_IDLE);](https://github.com/hathach/tinyusb/blob/7004914d8c6c7d88908776eb153661b6b987e513/src/host/usbh.c#L518)
fails because _ctrl_xfer.stage is equal to CONTROL_STAGE_ACK.

There is only 1 control transfer is active at a time, if it failed, try again later. You can implement your own fifo to queue up the pending control transfer.

closed 2023-03-22 by hathach

jfedor2, 2023-04-08:
No, you can't just try again later. It is stuck in ACK stage forever. All subsequent attempts fail.