hathach/tinyusb #1833: Eliminate compiler warning.

clhenry, 2023-01-01:
### Related area

FreeRTOS OSAL.

### Hardware specification

Custom MSP430F5638.

### Is your feature request related to a problem?

I'm getting an `integer conversion resulted in truncation` warning in `osal_queue_send()` from the third argument of  `xQueueSendToBack()` when using TI's CGT compiler version 21.6.1 LTS. The warning stems from the fact my device is a 16-bit system and my port is configured for 16-bit ticks using `configUSE_16_BIT_TICKS` but `OSAL_TIMEOUT_WAIT_FOREVER` is defined as `UINT32_MAX`.

### Describe the solution you'd like

I'm recommending changing `xQueueSendToBack(qhdl, data, OSAL_TIMEOUT_WAIT_FOREVER)` to `xQueueSendToBack(qhdl, data, _osal_ms2tick((OSAL_TIMEOUT_WAIT_FOREVER))` which resolves the 3rd argument to `portMAX_DELAY`.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-01-06:
this is not suitable for a feature request

clhenry, 2023-01-07:
@hathach Any thoughts on this?

hathach, 2023-03-02:
sorry for late response, I don't have the setup to test with, would you mind submitting a PR for this ?