hathach/tinyusb #1838: [Midi] Cable_num working for available and read

bjonnh, 2023-01-08:
### Related area

device stack

### Hardware specification

rp2040,generic

### Is your feature request related to a problem?

Kind of.

I'm trying to use tinyusb from adafruit tinyusb (which if I understand it correctly uses this project)

I made a wrapper class for the midi device but I would need two additional functions to manage cable_num:
tud_midi_n_available
tud_midi_n_stream_read

They take cable_num as arguments but do nothing with them.

### Describe the solution you'd like

A support for cable_num for those two methods.

That's the only thing missing for being able to map USB with multiple jacks to fortyseveneffects/MIDI Library.

I may be able to do a PR for it, but I will need to dig deeper in this code.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-01-13:
please make an PR for this if possible, thanks.

rppicomidi, 2023-04-18:
@bjonnh Did you come up with a good solution for this issue? My solution to this problem in my projects that deal with multiple cables on input was to use raw MIDI packets (`tud_midi_packet_read()`) and manage the cable number in the application.

bjonnh, 2023-04-18:
I have a solution yes, I'm just not sure how this could be included in tinyusb.

You can find the code here for a sort of "generic driver"
https://github.com/bjonnh/dinoctopus/tree/main/lib/usbmidi

I'm using the AdaFruit library and I'm storing the cable number. 

This is how I use it:
https://github.com/bjonnh/dinoctopus/blob/main/src/pico/midi/midirouter.cpp

I'm just not sure yet how to include that properly in tiny usb.

aspdigital, 2023-04-18:
I did USB-MIDI for SiLabs EFM8UB and EFM32GG, using the SiLabs USB stack, and I ported it to TI Tiva. This was before I learned of TinyUSB.

The issue, I think, is that so much of what a MIDI thing (controller, sample playback device, whatever) does is application dependent. This leaves the use of things like Cable Numbers entirely undefined at the protocol level. It is up to the application to look at the (if MIDI In) incoming packet and decide if the Cable Number, the Channel, and even the specific controller is relevant to what you're doing.

As an example, see [here](https://github.com/aspdigital/TM4C-USBSerialMIDI). This is an example of using USB-MIDI on the TI Tiva. There are four MIDI interfaces. Two are IN to the computer, two are OUT of the computer. These are handled by Cable Numbers, which is a way of saying MIDI Port (which is how macOS considers them). 

Cable 0 OUT from the computer controls the RGB LED on the eval board. Send the value on the correct controller and the LED changes.

Cable 1 OUT from the computer drives a UART for sending MIDI messages out on the usual DIN connector.

Cable 0 IN to the computer gets button presses as Note On controllers.

Cable 1 IN to the computer takes what comes in on a serial DIN jack and sends it to the computer without interpreting it.

Messages from the computer to the MCU that did not target the specific Cable, Channel and Controller are simply dropped on the floor. In the other direction, the MCU will not send messages except on a specific Cable, Channel and Controller.

The whole point is that this stuff is really application dependent.

rppicomidi, 2023-04-18:
I looked at making the two function calls in tinyusb work as the API defines them. If we keep the USB MIDI receive FIFO structure the same, the `tud_midi_n_available()` function requires peaking at the packets in the USB receive FIFO without pulling data and counting the number of packets that match the specified cable number. The `tud_midi_n_stream_read()` function would require pulling only those packets from the specified cable out of the receive FIFO, which is possible but complicated. An alternative implementation would be to have up to 16 different receive FIFOs; one per possible MIDI cable. That is consistent with treating each cable number as a separate MIDI port and is easier to implement. However, to keep streams in proper time sequence across FIFOs, packets would need to be time stamped. Adding a time stamp to each packet plus making the number of FIFOs match the number of cables could start to become a memory problem on some systems.

A simpler API that uses current infrastructure without much modification would be if the `cable_num` argument of a `tud_midi_n_stream_read()`-like function were a pointer to the cable number. The function would read from packets from the USB receive buffer until the cable number changes. The function would still return the number of bytes in the stream, but it would also set `*cable_num` to the cable number in the stream. An application would call the function until it returns 0 and route the received MIDI stream in the application. For example,
```
// Return the number of bytes read in the stream and set *cable_num to the cable number in the stream.
// Return 0 when when there are no more streams or stream fragments in the receive FIFO
// If cable_num is NULL, then this function behaves like to tud_midi_stream_read()
uint32_t tud_midi_demux_stream_read  (uint8_t* cable_num, void* buffer, uint32_t bufsize);
```
@hathach I can submit a pull request to implement the `tud_midi_demux_stream_read()` function if you would like.