hathach/tinyusb #1850: [Teensy 4.0 host] devices don't work if pre-plugged

jfedor2, 2023-01-12:
### Operating System

Linux

### Board

Teensy 4.0

### Firmware

examples/dual/host_hid_to_device_cdc

### What happened ?

I'm using a Teensy 4.0 with a USB port soldered to the host pads. The example works if I first connect the Teensy to a computer and then plug a mouse to the host port on the Teensy. But if I plug a mouse to the Teensy first and then connect the Teensy to a computer, it doesn't work.

With the change below applied, it works. (I'm not saying this is the proper fix, just to help localize the issue I'm experiencing.)

```
diff --git a/src/portable/ehci/ehci.c b/src/portable/ehci/ehci.c
index 7140897a1..6f03242f3 100644
--- a/src/portable/ehci/ehci.c
+++ b/src/portable/ehci/ehci.c
@@ -238,6 +238,8 @@ void hcd_device_close(uint8_t rhport, uint8_t dev_addr)
   ehci_data.regs->command_bm.async_adv_doorbell = 1;
 }
 
+static void port_connect_status_change_isr(uint8_t rhport);
+
 bool ehci_init(uint8_t rhport, uint32_t capability_reg, uint32_t operatial_reg)
 {
   (void) capability_reg; // not used yet
@@ -248,6 +250,8 @@ bool ehci_init(uint8_t rhport, uint32_t capability_reg, uint32_t operatial_reg)
 
   ehci_registers_t* regs = ehci_data.regs;
 
+  uint32_t connect_status_change = regs->portsc_bm.connect_status_change;
+
   //------------- CTRLDSSEGMENT Register (skip) -------------//
   //------------- USB INT Register -------------//
   regs->inten  = 0;                 // 1. disable all the interrupt
@@ -318,6 +322,11 @@ bool ehci_init(uint8_t rhport, uint32_t capability_reg, uint32_t operatial_reg)
   //------------- ConfigFlag Register (skip) -------------//
   regs->portsc_bm.port_power = 1; // enable port power
 
+  if (connect_status_change)
+  {
+    port_connect_status_change_isr(rhport);
+  }
+
   return true;
 }
```

### How to reproduce ?

1. Flash a Teensy 4.0 board with the host_hid_to_device_cdc example.
2. Disconnect the Teensy from the computer.
3. Plug a mouse to the host port on the Teensy.
4. Connect the Teensy to a computer.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-05-31:
may be fixed by recent ehci PRs. Please try testing it again with master.

jfedor2, 2023-05-31:
Just tested on current master, it seems to be fixed. Thank you!