hathach/tinyusb #1851: [Teensy 4.0 host] Crash when unplugging a device

jfedor2, 2023-01-12:
### Operating System

Linux

### Board

Teensy 4.0

### Firmware

examples/dual/host_hid_to_device_cdc

### What happened ?

I'm using a Teensy 4.0 board with a USB port soldered to the host pads. Plugging in a mouse and then unplugging it causes the board to crash (failing assert in usbh.c:481). If I comment out the assert, everything seems to work OK (I can plug the device back and it works).

### How to reproduce ?

1. Flash a Teensy 4.0 board with the host_hid_to_device_cdc example.
2. Plug in a mouse to the host USB port on the Teensy.
3. Unplug the mouse.

### Debug Log as txt file

```
regs->portsc = 1D00100A
  Current Connect Status: 0
  Connect Status Change : 1
  Port Enabled          : 0
  Port Enabled Change   : 1
  Port Reset            : 0
  Port Power            : 1
[1:0:0] USBH DEVICE REMOVED
  Address = 1
HID close
HUB close
on EP 81 with 0 bytes
tuh_task_ext 481: ASSERT FAILED
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Ryzee119, 2023-01-12:
Might be related to https://github.com/hathach/tinyusb/issues/1604

hathach, 2023-01-12:
thanks for reporting the issue, look like the usb transfer is complete -> device is unplugged --> usbh_task() is called and it asserts since it couldn't find the owner of the usb transfer (is unplugged now). Can you add the following log right above the assert and paste the log output here. I will pull out hardware and do more test later on.

```C
                // no driver/callback responsible for this transfer
                TU_LOG2_INT(tuh_mounted(event.dev_addr));
                TU_ASSERT(false, );
```

jfedor2, 2023-01-12:
Here's what it says when I add that log and unplug:

```
regs->portsc = 1D00100A
  Current Connect Status: 0
  Connect Status Change : 1
  Port Enabled          : 0
  Port Enabled Change   : 1
  Port Reset            : 0
  Port Power            : 1
[1:0:0] USBH DEVICE REMOVED
  Address = 1
HID close
HUB close
on EP 81 with 0 bytes
tuh_mounted(event.dev_addr) = 0
tuh_task_ext 482: ASSERT FAILED
```

hathach, 2023-01-13:
thanks that confirm my suspect, I will pull out my hw and fix it later when possible.

hathach, 2023-05-31:
this should be fixed in recent imx usb host improvement PR. Would you mind testing it again with master.

closed 2023-05-31 by jfedor2