hathach/tinyusb #1855: Add static analysis reporting

silvergasp, 2023-01-13:
### Related area

Security

### Hardware specification

N/A

### Is your feature request related to a problem?

Beyond compiler warnings/errors tinyusb doesn't currently leverage static analysis to prevent bugs.

### Describe the solution you'd like

 I'd like to propose integrating [clang-tidy](https://clang.llvm.org/extra/clang-tidy/) into the continuous integration workflow. It's worth noting that clang-tidy is also capable of many other things statically like;
- Identifying bugprone coding patterns.
- Identifying usage of insecure functions.
- Identifying performance problems.
- Identifying readibility issues.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-01-17:
Superb, I have tried a couple of static analyzer before, but didn't got anywhere. This look great, if you could make an PR for this, I am happy to review. Otherwise, I will try to wrap my head around this later on. The more checks we run, the better :)

silvergasp, 2023-01-17:
Yeah clang-tidy is a nice easy one to use. In fact, I've already run it on tinyusb. Just go a bit of cleanup to do before a PR. Though I do have a question. Is there a way to build all of the source code under `/src` minus all the ports, in one make invocation? I've been using [bear](https://github.com/rizsotto/Bear) (a command line tool) to generate a `compile-commands.json` file which is used for clang-tidy but also clangd, but I need to be able to build all of tinyusb in one `make` invocation and I'm having trouble wrapping my head around how to do that. I'd like to build;
- src/common
- src/classes
- src/*.c

But not any of the portable backends.

hathach, 2023-01-17:
what would be the issue you having, is it due to SYMBOL check e.g `CFG_TUD_CDC` that opt-out the code. We can add tusb_config.h or CFLAGS to enable all the feature. For the backend e.g dcd_*, we can use the port/template one, which is basically an stub for backend APIs. 

I am not familiar with clang-tidy, though if you share your working, maybe I could help it out (not entirely sure though).

silvergasp, 2023-01-17:
Hmm I think the port/template that you mentioned, is the missing piece I was looking for. I'll try and use that backend for static analysis, and see where that gets me.

hathach, 2025-10-27:
resovled by https://github.com/hathach/tinyusb/pull/3318