hathach/tinyusb #1874: Issue with CDC USB Serial to Arduino Due

phatchman, 2023-01-29:
### Operating System

RaspberryPi OS

### Board

RP2040

### Firmware

examples/host/cdc_msc_hid

### What happened ?

I am trying to use a Pico as USB Host to and Arudino Due for a USB Serial Connection. When data is sent, an assertion fails. While the below says the assertion fails on line 484, it is actually line 481. The difference is due to some additional logging statements added by me.

### How to reproduce ?

Build examples/host/cdc_msc_hid on pico and load onto pico.
Connect usb cable from Arudino Due to pico
send data from Due to Pico

### Debug Log as txt file

þTinyUSB Host CDC MSC HID Example
USBH init on controller 0
sizeof(usbh_device_t) = 304
sizeof(hcd_event_t) = 12
sizeof(_ctrl_xfer) = 24
sizeof(tuh_xfer_t) = 24
sizeof(tu_fifo_t) = 20
sizeof(tu_edpt_stream_t) = 40
CDC init
MSC init
HID init
HUB init
sizeof(hw_endpoint_t) = 36
.[0:] USBH DEVICE ATTACH
Full Speed
[0:0] Open EP0 with Size = 8
Returning 1
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
Transfer complete
on EP 00 with 8 bytes
Buffer complete
Transfer complete
on EP 80 with 8 bytes
[0:0] Control data:
  0000:  12 01 10 01 02 00 00 08                          |........|
Buffer complete
Transfer complete
on EP 00 with 0 bytes

Set Address = 1
[0:0] Set Address: 00 05 01 00 00 00 00 00
Transfer complete
on EP 00 with 8 bytes
Buffer complete
Transfer complete
on EP 80 with 0 bytes

[0:1] Open EP0 with Size = 8
Returning 1
Get Device Descriptor
[0:1] Get Descriptor: 80 06 00 01 00 00 12 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Buffer complete
Transfer complete
on EP 80 with 18 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4
[0:1] Control data:
  0000:  12 01 10 01 02 00 00 08 41 23 3D 00 01 00 01 02  |........A#=.....|
  0010:  DC 01                                            |..|
Buffer complete
Transfer complete
on EP 00 with 0 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4

Get Configuration[0] Descriptor (9 bytes)
[0:1] Get Descriptor: 80 06 00 02 00 00 09 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Transfer complete
on EP 80 with 9 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4
[0:1] Control data:
  0000:  09 02 3E 00 02 01 00 C0 32                       |..>.....2|
Buffer complete
Transfer complete
on EP 00 with 0 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4

Get Configuration[0] Descriptor
[0:1] Get Descriptor: 80 06 00 02 00 00 3E 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Buffer complete
Buffer complete
Buffer complete
Transfer complete
on EP 80 with 62 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4
[0:1] Control data:
  0000:  09 02 3E 00 02 01 00 C0 32 09 04 00 00 01 02 02  |..>.....2.......|
  0010:  01 00 05 24 00 01 10 04 24 02 06 05 24 06 00 01  |...$....$...$...|
  0020:  07 05 82 03 08 00 FF 09 04 01 00 02 0A 00 00 00  |................|
  0030:  07 05 04 02 40 00 01 07 05 83 02 40 00 01        |....@......@..|
Buffer complete
Transfer complete
on EP 00 with 0 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4

Parsing Configuration descriptor (wTotalLength = 62)
opening CDC
  Open EP 82 with Size = 8
Allocate interrupt ep 0
  Open EP 04 with Size = 64
Allocate bulk ep 1
  Open EP 83 with Size = 64
Allocate bulk ep 2
  CDC opened
TUSB_DEC_ENDPOINT = 5, tu_desc_type = 4, ep_addr = 00
TUSB_DEC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 00
TUSB_DEC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 02
TUSB_DEC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 06
TUSB_DEC_ENDPOINT = 5, tu_desc_type = 5, ep_addr = 82
  Bind EP 82 to driver id 0
opening CDC
opening MSC
opening HID
opening HUB
Set Configuration = 1
[0:1] Set Configuration: 00 09 01 00 00 00 00 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Transfer complete
on EP 80 with 0 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4

Device configured
CDC set config: itf = 0
CDC Set Control Line State
[0:1] Class Request: 21 22 03 00 00 00 00 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Transfer complete
on EP 80 with 0 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4

CDC Set Line Conding
[0:1] Class Request: 21 20 00 00 00 00 07 00
Transfer complete
on EP 00 with 8 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
Buffer complete
Transfer complete
on EP 00 with 7 bytes
epnum = 0, ep_addr = 0, DRIVER_COUNT = 4
[0:1] Control data:
  0000:  00 C2 01 00 00 00 08                             |.......|
Buffer complete
Transfer complete
on EP 80 with 0 bytes
epnum = 0, ep_addr = 80, DRIVER_COUNT = 4

CDC Interface is mounted: address = 1, itf_num = 0
  Baudrate: 115200, Stop Bits : 0
  Parity  : 0, Data Width: 8
  Queue EP 83 with 64 bytes ... OK
A device with address 1 is mounted
Buffer complete
on EP 83 with 44 bytes
epnum = 3, ep_addr = 83, DRIVER_COUNT = 4
drv_id = 255, epnum = 3, epdir = 1
CFG_TUH_API_EDPT_XFER
tuh_task_ext 484: ASSERT FAILED

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

phatchman, 2023-01-30:
I got this "working" by adding the following to tu_edpt_bind_driver
    ep2drv[tu_edpt_number(0x83)][tu_edpt_dir(0x83)] = driver_id;
    ep2drv[tu_edpt_number(0x04)][tu_edpt_dir(0x04)] = driver_id;

Obviously this isn't the correct fix, but at least gives me something to work with for now.

I can see that endpoint addresses 04 and 83 aren't being found by this function, even though they are returned during the parsing of the configuration descriptor.
TUSB_DESC_ENDPOINT = 5, tu_desc_type = 4, ep_addr = 00
TUSB_DESC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 00
TUSB_DESC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 02
TUSB_DESC_ENDPOINT = 5, tu_desc_type = 36, ep_addr = 06
TUSB_DESC_ENDPOINT = 5, tu_desc_type = 5, ep_addr = 82

hathach, 2023-01-31:
please put the debug log into txt file for readability

hathach, 2023-01-31:
this should be fixed by #1877 please try to pull the PR changes to see if that works for you.

closed 2023-02-01 by hathach

phatchman, 2023-02-11:
Confirm fixed.

hathach, 2023-02-11:
Thanks for your confirmation

PR #1877 (2023-02-01): Fix host cdc without iad
**Describe the PR**
- fix #1874  host issue when mounting an CDC device without IAD
- also fix #1859 
- add fuzzing status badge