hathach/tinyusb #1879: Wait forever mutex leading to unrecoverable hangups

MWP, 2023-01-31:
### Operating System

Others

### Board

STM32H7 Custom 

### Firmware

STM32H7 using TinyUSB github master pulled on the 2022-05-24.
FreeRTOS 202210.00 .
Main use is two CDC ports.

### What happened ?

I'm having occasional random problems with tu_edpt_claim() stalling due to the passed mutex never being unlocked.
tu_edpt_claim() is being called up the chain from a tud_cdc_n_write_flush() in a FreeRTOS static task with low priority.
tud_task() is run from a task with the highest priority.

Obviously there is a problem else where causing this, but why do mutex's in TinyUSB ever use a forever wait time?
Surely there is an upper reasonable limit that could be used to prevent hangups like this?

### How to reproduce ?

Running my firmware. Sorry, cant provide much more detailed information than this.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-02-01:
closed firmware is not reproducible and therefore not trouble-shootable. If you think the issue is caused by the mutex, please try to provide an simplest example that could reproduce the issue.

closed 2023-02-01 by hathach

MWP, 2023-02-01:
> closed firmware is not reproducible and therefore not trouble-shootable. If you think the issue is caused by the mutex, please try to provide an simplest example that could reproduce the issue.

This shouldn't be closed yet. I can troubleshoot it with some help.

At some point in TinyUSB, the mutex is not being released, so on the next attempt to lock it, the lock function never returns as it's called with no-timeout (forever).

There are two problems here:
1 - the mutex is not released somewhere (i dont know how to track this down, any advice?)
2 - tinyusb locks mutexes with a wait forever as the timeout value (why is a forever value used?)

hathach, 2023-02-01:
reproducible code/example is required for a bug template, if you want to discuss/question please use the discussion ticket instead.