hathach/tinyusb #1892: bit field arrangement is not guaranteed by the C standard (cdc.h)

jefftrull, 2023-02-10:
### Operating System

Linux

### Board

n/a

### Firmware

examples/device/cdc_msc

### What happened ?

Compilation under sdcc (4.0 and 4.2) fails on the assert at line 414 of cdc.h which checks the size of `cdc_line_control_state_t`. The exact layout of bitfields is apparently implementation-defined and sdcc makes a different choice than gcc in this case. See [the SDCC discussion](https://sourceforge.net/p/sdcc/bugs/3530/) for more info. So this seems to be a kind of C standard conformance issue.

There is a straightforward workaround which I will supply in a PR.

### How to reproduce ?

Build tinyusb under sdcc for 8051, for example by using [my branch](https://github.com/jefftrull/tinyusb/tree/mcs51). Change src/class/cdc/cdc.h to match master and observe this error message:

`src/class/cdc/cdc.h:410: warning 215: static assertion failed: "size is not correct"`

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

jefftrull, 2023-02-10:
sdcc also rejects bitfields larger than 16 bits; the PR will cover that as well

hathach, 2023-02-13:
can you specify which board/mcu you are using, if possible please add an PR to include the mcu into board support, so that we could include the SDCC into ci build to fix any future changes.

jefftrull, 2023-02-13:
This will eventually be for the (obsolete) ALi M5623 in certain Canon scanners; I am in the process of porting to SDCC but am facing code size issues that require significant changes to the TinyUSB core so a PR is somewhat in the future.

hathach, 2023-02-13:
thank you for the info, indeed the code size has been increased quite a bit lately due to lots of added features. Maybe in the future we should have some option to trim down the code size e.g no dynamic configuration etc..

closed 2023-02-13 by hathach