hathach/tinyusb #1897: AUDIO examples fail for me, while CDC and MIDI work

woolfman007, 2023-02-12:
### Operating System

MacOS

### Board

Nucleo F413ZH

### Firmware

examples/audio_test

### What happened ?

I added tinyusb to my CubeIDE project (which worked mostly flawlessly). Then I tried some of the examples. Both audio examples refused to work, while cdc_dual_ports and midi_test worked without any problems. I have another project, where I use STM's default USB middleware and there I can implement a working USB audio device; I also copied over the clock settings from this project in order to make sure they work. It may not be 1:1 comparable, since the audio_test example is a microphone, whereas my STM project defines a speaker.

When plugging into Windows, device manager recognizes the device as a microphone, but complains that it cannot be started. 

### How to reproduce ?

1. Create a project in CubeIDE
2. activate USB_OTG, enable global USB interrupt 
3. fix clock settings
4. add tinyusb/src to the project  
5. exclude src/portable/synopsis from build
6. set CFG_TUSB_MCU=OPT_MCU_STM32F4
7. copy over and add tusb_config.h and usb_descriptors.c from the various examples
8. I had to add some custom defines to tusb_config.h;
      ```
      #define BOARD_DEVICE_RHPORT_SPEED   OPT_MODE_FULL_SPEED
      #define CFG_TUSB_RHPORT0_MODE      (OPT_MODE_DEVICE | BOARD_DEVICE_RHPORT_SPEED)
      #define CFG_TUSB_RHPORT1_MODE (OPT_MODE_NONE)
```
9. add a call to `tusb_init();` and hook up `tud_int_handler(0);`

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

woolfman007, 2023-02-12:
after further investigation I think this is related to the fact that the Nucleo onboard OTG port is only capable of USB FS, wheres USB AUDIO 2.0 requires HS, if I'm not mistaken

HiFiPhile, 2023-02-12:
> wheres USB AUDIO 2.0 requires HS, if I'm not mistaken

It's not case.

Could you try #1825 ?

woolfman007, 2023-02-12:
I tried it, but unfortunately, this does not fix my issue

HiFiPhile, 2023-02-12:
Please enable debug logging in tusb_config.h and upload log.

You can also try F412 BSP in https://github.com/hathach/tinyusb/tree/master/hw/bsp/stm32f4/boards/stm32f412nucleo
normally they have the same USB IP.

woolfman007, 2023-02-12:
Thanks for coming back to me! 

I set debug level to 3 and hooked up tu_printf to an UART. Here's what I got when plugging the device in:

```
USBD init on controller 0
sizeof(usbd_device_t) = 43
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 24
AUDIO init
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 09 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 FF 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 13 bytes ...
USBD Xfer Complete on EP 80 with 13 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 FF 00
  Get Descriptor String[3]
  Queue EP 80 with 14 bytes ...
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes ...
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 16 bytes ...
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
[... 123 lines cut ...]
  Stall EP0

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 16 bytes ...
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 16 bytes ...
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 16 bytes ...
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Suspend : Remote Wakeup = 0

```

HiFiPhile, 2023-02-12:
Please use unmodified example, your are missing tud_audio_get_req_entity_cb() in exaplem's main.c

woolfman007, 2023-02-12:
oh sorry for the confusion... I forgot to re-enable the resp. source file. But that is not the issue, it does not work either way. I updated the log above to what's being reported when `tud_audio_get_req_entity_cb()` and the other callbacks are enabled. I also checked with a breakpoint and the said function was definitely called when I collected the updated log data above

woolfman007, 2023-02-12:
I tried the UAC2 headset example and this seems to work (at least partly). 

I got two runtime assertions in dcd_synopsis.c; 

- in handle_rxflvl_ints(), where the switch went to the default branch 
- in dcd_edpt_open(), where `TU_ASSERT(sz + _allocated_fifo_words_tx <= EP_FIFO_SIZE/4);` failed

but I can stream audio to the device and record it back :-) The recorded audio is a bit corrupted, which I'll investigate now. But still this is a huge step forward for me

woolfman007, 2023-02-12:
I fact I get loads of assertions with the headset example... I took a chance and commented some of them out just to see if it works in principle; however, finnaly I got this:

```
          usbd_class_driver_t const * driver = get_driver( _usbd_dev.ep2drv[epnum][ep_dir] );
          TU_ASSERT(driver, );

```
in usbd.c, which of course cannot be skipped...

HiFiPhile, 2023-02-12:
No you touch asserts they exists for a reason !
If you got USBD asserts their is something seriously wrong in your side.

You said you excluded dcd_synopsis.c (which is good to use dwc2.c) how did you hit assert inside it?

woolfman007, 2023-02-12:
there are two synopsis folders, one `portable/st/synopsis` and one `portable/synopsis`. The latter seems to be deprecated (at least that's what I read in another report) and would generate compile errors and / or duplicate symbols for me. 

And, I would never leave the assertions commented out, I just wanted to see if it works at all. I know they're there for a reason :-)