hathach/tinyusb #1915: arm-none-eabi-gcc: No such file or directory

Bexin3, 2023-02-19:
### Operating System

MacOS

### Board

Seeding Xiao

### Firmware

examples/device/audio_test

### What happened ?

make: arm-none-eabi-gcc: No such file or directory
make: *** [_build/seeeduino_xiao/obj/src/portable/microchip/samd/dcd_samd.o] Error 1

### How to reproduce ?

Try to compile.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2023-02-19:
You should install the toolchain and add to PATH before compiling.

closed 2023-02-19 by HiFiPhile

Bexin3, 2023-02-19:
Hi, how would I do this? I followed the Quickstart guide, and its not working, and I am kinda new to using command console.

aspdigital, 2023-02-20:
How do you add the tools to the path? 
Current Macs use the Z Shell, so in your home directory there's a file called .zshrc (leading period is important). Edit that file to add the path to wherever the installer put the tools. Tho I think if you installed the tools from the [official ARM package]https://developer.arm.com/-/media/Files/downloads/gnu/12.2.rel1/binrel/arm-gnu-toolchain-12.2.rel1-darwin-arm64-arm-none-eabi.pkg?rev=495058480373470f98133166d00ff0c2&hash=45A682B78E26256ACBB7F13A97219130(url) the path to the tools should be added to your path already.

Bexin3, 2023-02-24:
Hi, installed it from there but it didn't help. I have recently read the path recently changed, could that be the issue?

Bexin3, 2023-02-24:
realpath: illegal option -- -
usage: realpath [-q] [path ...]
CC dcd_samd.o
make: arm-none-eabi-gcc: No such file or directory
make: *** [_build/seeeduino_xiao/obj/src/portable/microchip/samd/dcd_samd.o] Error 1