hathach/tinyusb #1916: From Discussion 1886, "the examples in the package do not build" on macOS

aspdigital, 2023-02-20:
### Operating System

MacOS

### Board

lpcxpresso55s28

### Firmware

Any of the examples.

### What happened ?

See [here](https://github.com/hathach/tinyusb/discussions/1886) for details, but in a nutshell, the provided makefiles do not build the main.c and other files in the example source directory.

### How to reproduce ?

Clone the tinyusb project.
Change to the desired example project directory, e.g., /Users/andy/Projects/tinyusb/examples/device/midi_test

```
andy@mitzvah midi_test % make BOARD=lpcxpresso55s28 get-deps
andy@mitzvah midi_test % make BOARD=lpcxpresso55s28 all     
realpath: illegal option -- -
usage: realpath [-q] [path ...]
"PORT1 High Speed"
AS startup_LPC55S28_asm.o
CC dcd_lpc_ip3511.o
CC system_LPC55S28.o
CC fsl_clock.o
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c: In function 'CLOCK_EnableUsbhs0PhyPllClock':
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c:1986:56: warning: unused parameter 'src' [-Wunused-parameter]
 1986 | bool CLOCK_EnableUsbhs0PhyPllClock(clock_usb_phy_src_t src, uint32_t freq)
      |                                    ~~~~~~~~~~~~~~~~~~~~^~~
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c: In function 'CLOCK_EnableUsbhs0DeviceClock':
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c:2074:54: warning: unused parameter 'src' [-Wunused-parameter]
 2074 | bool CLOCK_EnableUsbhs0DeviceClock(clock_usbhs_src_t src, uint32_t freq)
      |                                    ~~~~~~~~~~~~~~~~~~^~~
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c:2074:68: warning: unused parameter 'freq' [-Wunused-parameter]
 2074 | bool CLOCK_EnableUsbhs0DeviceClock(clock_usbhs_src_t src, uint32_t freq)
      |                                                           ~~~~~~~~~^~~~
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c: In function 'CLOCK_EnableUsbhs0HostCloc':
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c:2086:52: warning: unused parameter 'src' [-Wunused-parameter]
 2086 | bool CLOCK_EnableUsbhs0HostClock(clock_usbhs_src_t src, uint32_t freq)
      |                                  ~~~~~~~~~~~~~~~~~~^~~
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/devices/LPC55S28/drivers/fsl_clock.c:2086:66: warning: unused parameter 'freq' [-Wunused-parameter]
 2086 | bool CLOCK_EnableUsbhs0HostClock(clock_usbhs_src_t src, uint32_t freq)
      |                                                         ~~~~~~~~~^~~~
CC fsl_power.o
CC fsl_reset.o
CC fsl_gpio.o
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/drivers/lpc_gpio/fsl_gpio.c: In function 'GPIO_EnablePortClock':
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/drivers/lpc_gpio/fsl_gpio.c:42:45: warning: unused parameter 'base' [-Wunused-parameter]
   42 | static void GPIO_EnablePortClock(GPIO_Type *base, uint32_t port)
      |                                  ~~~~~~~~~~~^~~~
CC fsl_common_arm.o
CC fsl_flexcomm.o
CC fsl_usart.o
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/drivers/flexcomm/fsl_usart.c: In function 'USART_TransferGetReceiveCount':
/Users/andy/Projects/tinyusb/hw/mcu/nxp/mcux-sdk/drivers/flexcomm/fsl_usart.c:992:52: warning: unused parameter 'base' [-Wunused-parameter]
  992 | status_t USART_TransferGetReceiveCount(USART_Type *base, usart_handle_t *handle, uint32_t *count)
      |                                        ~~~~~~~~~~~~^~~~
CC sct_neopixel.o
/Users/andy/Projects/tinyusb/lib/sct_neopixel/sct_neopixel.c: In function 'sctpix_init':
/Users/andy/Projects/tinyusb/lib/sct_neopixel/sct_neopixel.c:150:27: warning: unused parameter 'neoPixelType' [-Wunused-parameter]
  150 | void sctpix_init(uint32_t neoPixelType) {
      |                  ~~~~~~~~~^~~~~~~~~~~~
CC family.o
CC board.o
CC tusb.o
CC tusb_fifo.o
CC usbd.o
CC usbd_control.o
CC audio_device.o
CC cdc_device.o
CC dfu_device.o
CC dfu_rt_device.o
CC hid_device.o
CC midi_device.o
CC msc_device.o
CC ecm_rndis_device.o
CC ncm_device.o
CC usbtmc_device.o
CC video_device.o
CC vendor_device.o
LINK _build/lpcxpresso55s28/midi_test.elf
arm-none-eabi-gcc: error: _build/lpcxpresso55s28/obj//src/main.o: No such file or directory
arm-none-eabi-gcc: error: _build/lpcxpresso55s28/obj//src/usb_descriptors.o: No such file or directory
make: *** [_build/lpcxpresso55s28/midi_test.elf] Error 1

```

The fix is to modify the example makefile and change:
`SRC_C +=  $(addprefix $(CURRENT_PATH)/, $(EXAMPLE_SOURCE))`
to
`SRC_C += $(EXAMPLE_SOURCE)`

I do not know if this is a universal problem or if it is specific to macOS. Nor do I know if it is an artifact of building all of the examples at once.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

aspdigital, 2023-02-20:
As a follow-up, I tested this on a Windows 10 box and I got a bunch of different warnings but the files in the example src directory compiled and the whole thing linked.

There must be a difference between whatever make program I have on my Windows box and whatever is provided by Apple in Xcode. If you look at the top of the transcript posted above, you see:

`andy@mitzvah midi_test % make BOARD=lpcxpresso55s28 all`     
`realpath: illegal option -- -`
`usage: realpath [-q] [path ...]`

but this complaint is not thrown by the Windows make.

Bexin3, 2023-02-25:
Hi, I have the same issue, tried your fix but its still broken.

hathach, 2023-03-06:
should be fixed by https://github.com/hathach/tinyusb/pull/1938 , will merge shortly, check it out to see if that works for you

PR #1938 (2023-03-06): Fix build on windows and macos
**Describe the PR**
- fix #1916
- Use built-in make $(abspath) instead of shell realpath to be more compatible with windows and macos. 
- add windows and macos instance to run ci with only stm32f411disco board
closed 2023-03-06 by hathach