hathach/tinyusb #1926: Despite having it installed, I get arm-none-eabi-gcc: No such file or directory

Bexin3, 2023-02-24:
### Operating System

MacOS

### Board

Seeding Xiao

### Firmware

Audio_test

### What happened ?

realpath: illegal option -- r
usage: realpath [-q] [path ...]
CC dcd_samd.o
make: arm-none-eabi-gcc: No such file or directory
make: *** [_build/seeeduino_xiao/obj/src/portable/microchip/samd/dcd_samd.o] Error 1

### How to reproduce ?

Trying to make the example

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

aspdigital, 2023-02-25:
See the discussion [here](https://github.com/hathach/tinyusb/blob/master/docs/reference/getting_started.rst).

There are two issues here. The first is the error regarding `realpath`. These error appears when building on macOS. I didn't see it when building on a Windows machine using the GNU tools, especially make. This error hides the real problem.

The Makefile in each example directory has this line:

`SRC_C +=  $(addprefix $(CURRENT_PATH)/, $(EXAMPLE_SOURCE))`

and it is apparently the `addprefix` Makefile directive that fails and throws that error:

```
andy@mitzvah midi_test % make BOARD=lpcxpresso55s28 all     
realpath: illegal option -- -
usage: realpath [-q] [path ...]
```
This error causes all of the sources in the example directory (not TinyUSB itself) to not compile as the build system cannot find them. 

I don't know a proper fix for this, so I eliminated the `addprefix` line and substituted  
`SRC_C += $(EXAMPLE_SOURCE)`
and the examples I tried all built without issue, and they ran as expected.

Give this a try.

Bexin3, 2023-02-25:
Hi, I tried to do this, but unfortunately I am still getting the same error.

Bexin3, 2023-02-25:
It seems like make.mk in shared examples directory makes this call, didn't forgive out why does it not work too

hathach, 2023-03-06:
should be fixed by https://github.com/hathach/tinyusb/pull/1938, please give it a try