hathach/tinyusb #1939: CI Failures due to CH32V307 HAL

maxgerhardt, 2023-03-06:
### Operating System

Windows 10

### Board

Pico

### Firmware

None involved

### What happened ?

The inclusion of the CH32V307 hal in https://github.com/hathach/tinyusb/commit/949950e592d9a23a4b449784b0f8e7e274129318 has caused a CI failure in Github Actions due to excessively long filenames, see [here](https://github.com/maxgerhardt/platform-raspberrypi/actions/runs/4338148082/jobs/7574632312).

>error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'

### How to reproduce ?

`git clone --recursive https://github.com/hathach/tinyusb.git` on a Windows system that does not have Win32 Long Paths and/or git long paths activated.

### Debug Log as txt file

None needed.

### Screenshots

None needed.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

maxgerhardt, 2023-03-06:
I want to avoid having to do extreme Github Actions hacking by editing the registry, putting the *Github Actions Windows machine to sleep*, waking it up, and resuming work like in https://github.com/appveyor/ci/issues/3129#issuecomment-540777559. I don't think these folders ("Application") are needed in TinyUSB at all?

maxgerhardt, 2023-03-06:
I've managed to get CI running again by actually only needing to activate long paths in git in https://github.com/maxgerhardt/platform-raspberrypi/commit/8e7f3fcf79b69e5e24022fdb342e206d1d7cbc46.

```yaml
      - name: Enable Long Paths
        run: |
          if [ "$RUNNER_OS" == "Windows" ]; then
            git config --system core.longpaths true
          fi
        shell: bash
```

Still, if it can be avoided, I'd love that users need to not do these modifcations.

hathach, 2023-03-09:
I don’t see why it is a bug with tinyusb. Since you are working with rp2040, you shouldn’t clone other mcu dirver

closed 2023-03-09 by hathach

maxgerhardt, 2023-03-09:
Can you not consider cloning only a subtree of the CH32V307 HAL that excludes the application which is bursting through the max Windows path length?

hathach, 2023-03-09:
I would love to include a minimal mcu driver that is used by others and well maintained if you could suggest one.

maxgerhardt, 2023-03-09:
I do host https://github.com/Community-PIO-CH32V/framework-wch-noneos-sdk which I always try to keep up-to-date. It contains the HAL for all CH32VXXX chips. No application code.

hathach, 2023-03-09:
Thanks, let me check this out later. Maybe I should just get rid of submodules by implementing #1464

hathach, 2023-03-11:
all submodules are removed now and replaced by an python script to manage dependency repos https://github.com/hathach/tinyusb/pull/1947

minisbett, 2023-03-24:
This issue does not seem to be fixed. I am using earlephilhower's arduino-pico framework and updated it's pico-sdk to use the latest state of tinyusb. The issue still exists.
```
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32u5xx_hal_driver': checked out '2e1d4cdb386e33391cb261dfff4fefa92e4aa35a'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32wbxx_hal_driver': checked out '2c5f06638be516c1b772f768456ba637f077bac8'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long

fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'

From https://github.com/ARM-software/CMSIS_5
 * branch                20285262657d1b482d132d20d755c8c330d55c1f -> FETCH_HEAD

Submodule path 'pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out '20285262657d1b482d132d20d755c8c330d55c1f'
Submodule path 'pico-sdk/lib/tinyusb/lib/FreeRTOS-Kernel': checked out '2a604f4a2818b8354b5e1a39e388eb5e16cfbc1f'
From https://github.com/lwip-tcpip/lwip
 * branch              159e31b689577dbf69cf0683bbaffbd71fa5ee10 -> FETCH_HEAD

Submodule path 'pico-sdk/lib/tinyusb/lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'pico-sdk/lib/tinyusb/lib/sct_neopixel': checked out 'e73e04ca63495672d955f9268e003cffe168fcd8'
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2': checked out '19615407727073e36d81bf239c52108ba92e7660'
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi'

Cloning into 'C:/Users/Niklas/.platformio/.cache/tmp/pkg-installing-wcupmhjt/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...

Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

fatal: Failed to recurse into submodule path 'pico-sdk'

From https://github.com/earlephilhower/bearssl-esp8266
 * branch            f294aa022f82833941ffdf3b0f306c2072e3a26c -> FETCH_HEAD

Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/minipadKB/arduino-pico.git', 'C:\\Users\\Niklas\\.platformio\\.cache\\tmp\\pkg-installing-wcupmhjt']
```

hathach, 2023-04-03:
@maxgerhardt I started to use PIO for some sketch recently, it is indeed a pain to wait for all submodules clone. I am glad we finally remove them in tinyusb project.