hathach/tinyusb #1944: `remove_volatile_cast()` not defined in src/portable/raspberrypi/rp2040/dcd_rp2040.c

dhalbert, 2023-03-09:
### Operating System

Linux

### Board

RP2040

### Firmware

latest `master`, as of today

### What happened ?

I'm bringing CircuitPython's use of tinyusb up to date. We were about a year behind. I'm getting a compile error here because the macro `remove_volatile_cast()` is not defined here:
https://github.com/hathach/tinyusb/blob/990fb6ae5c4d9d4b77c5a9ecb3a2abe899dd2712/src/portable/raspberrypi/rp2040/dcd_rp2040.c#L300
It is defined in `pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h`, but there doesn't seem to be an `#include hardware/sync.h` directly or indirectly done in `dcd_rp2040.c`.

See sample compile failure here:
https://github.com/adafruit/circuitpython/actions/runs/4379438712/jobs/7665918999#step:10:28

### How to reproduce ?

see above

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-03-10:
have you tried to bump up pico-sdk ? this cast is added recently by rpi team, and actually pass tinyusb ci.

dhalbert, 2023-03-10:
Yes, I am running with pico-sdk 1.5.0. `remove_volatile_cast()` is there, it's just that the .h file that defines it is not being included directly or indirectly by `dcd_rp2040.c`.

dhalbert, 2023-03-10:
I am thinking that `#include hardware/sync.h` should explicitly be in `dcd_rp2040.c`.

kilograham, 2023-03-10:
yes it probably should;

I assume this is failing because you aren't using OSAL_PICO

dhalbert, 2023-03-10:
> I assume this is failing because you aren't using OSAL_PICO

That's true. Maybe we should look at using osal across our multiple ports.

PR #1948 (2023-03-11): rp2040: include hardware/sync.h explicitly
- Fixes #1944.

https://github.com/hathach/tinyusb/blob/990fb6ae5c4d9d4b77c5a9ecb3a2abe899dd2712/src/portable/raspberrypi/rp2040/dcd_rp2040.c#L300
`dcd_rp2040.c` uses `remove_volatile_cast()`, which is defined in the `pico-sdk` in `hardware/sync.h`. But `dcd_rp2040.c` does not `#include hardware/sync.h`. When not using the `pico-sdk` build environment and `OSAL_PICO`, `remove_volatile_cast()` will not be defined. This PR adds an explicit `#include` to fix that.
closed 2023-03-11 by hathach