hathach/tinyusb #1945: CFG_TUSB_RHPORT1_MODE not defined in STM build

dhalbert, 2023-03-09:
### Operating System

Linux

### Board

RP2040

### Firmware

latest `master`, as of today

### What happened ?

I'm bringing CircuitPython's use of tinyusb up to date. We were about a year behind. Compile error because `CFG_TUSB_RHPORT1_MODE` is not defined. This used to work, but I think that #1350 may have changed things.
https://github.com/hathach/tinyusb/blob/990fb6ae5c4d9d4b77c5a9ecb3a2abe899dd2712/src/portable/st/synopsys/dcd_synopsys.c#L528
The statement above is not guarded with `#if` against `CFG_TUSB_RHPORT1_MODE` not being defined.
Example failure: https://github.com/adafruit/circuitpython/actions/runs/4379438712/jobs/7665911868#step:10:56

FYI, we define `CFG_TUSB_RHPORT1_MODE` conditionally here:
https://github.com/adafruit/circuitpython/blob/353411af4b384ecc595215e9ce1ae95153c4cac1/supervisor/shared/usb/tusb_config.h#L67-L79

Would it be better to do that a different way now?

circuitpython PR is https://github.com/adafruit/circuitpython/pull/7702

### How to reproduce ?

see above

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-03-10:
@dhalbert you need to switch to the new dwc2 for stm32 since the one in st/synopsys will be removed shortly in the future. Maybe that would fix the compiled issue.
- https://github.com/hathach/tinyusb/tree/master/src/portable/synopsys/dwc2

dhalbert, 2023-03-13:
`dwc2` builds and works, so I'll close this as unneeded since `st/synopsys` is going away.

closed 2023-03-13 by dhalbert

hathach, 2023-03-13:
glad that works out for you, let me know if stm32 port has any issues with the new driver