hathach/tinyusb #1949: uac2_headset on STM32F401 - TU_BREAKPOINT() on USB reconnect

elagil, 2023-03-10:
### Operating System

Windows 11

### Board

STM32F401 custom design

### Firmware

Same as https://github.com/hathach/tinyusb/issues/1935
(CubeMX Makefile project on FREERTOS, using the new `dcd_dwc2` driver)

### What happened ?

In general, I can connect the USB device and I can play music or loop it back to the microphone without issues. The looped back signal looks like the one I transmitted.

Now, there are two cases:
1. I pause/stop the music, disconnect USB, reconnect USB -> **no problems**
2. I leave the music playing, disconnect USB, reconnect USB -> **I hit `TU_BREAKPOINT()`** in `dcd_dwc2.c:1085` with `pktsts == 0` `pktsts == 9` or `pktsts == 11` (random).

Why could that happen, only when I reconnect when I previously had music playing?
I do hit both the `tud_umount_cb()` when disconnecting, and the `tud_mount_cb()` when reconnecting, before I run into the breakpoint.

My `tusb_config.h` has `__RX__` defined, so I limit the quality to 16 bit/48 kHz.

### How to reproduce ?

See above (point 2.)

### Debug Log as txt file

The log starts just before I disconnect (while playing music) and then reconnect.
A lot of the configuration goes well, but it fails at getting the set volume.

[audio_log.txt](https://github.com/hathach/tinyusb/files/10946693/audio_log.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

gabChouin, 2023-11-23:
I've hit the same issue with an STM32U575. I had similar behavior to what was described in https://github.com/hathach/tinyusb/issues/1935 but https://github.com/hathach/tinyusb/pull/1928/ allowed me to advance one step forward to this current bug (with the default settings of 96 kHz 24-bit). Do you have any updates on this issue ?

elagil, 2023-11-23:
@gabChouin Unfortunately not. I gave up eventually and wrote my own UAC (1.0 only) driver.

HiFiPhile, 2023-11-23:
Could you take a looks the version of dwc2 IP used in F401 ?
Basically these registers : https://github.com/hathach/tinyusb/blob/d6b8b3515df4bb83b6038bd60260771e65af8de7/src/portable/synopsys/dwc2/dwc2_info.py#L6C4-L6C4

I'll see if it's reproducible on my F429 board.

HiFiPhile, 2023-11-23:
Could you try #2333 ?

While I didn't hit breakpoint, sometimes after re-plug it NAK all OUT transfers (which will cause host to re-enumerate like in your log)

rpascalsdl, 2023-11-26:
I believe I have the same issue on RP2040. Don't have exact steps to reproduce, as I couldn't find an exact pattern when it happens. Here's an example of the problem when I was plugging and unplugging the Rasbperry Pi Pico from a Pixel 8 phone (flushed with uac2_headset example):

```
_exit(int status) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\src\rp2_common\pico_runtime\runtime.c:187)
panic(const char * fmt) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\src\rp2_common\pico_runtime\runtime.c:325)
_hw_endpoint_buffer_control_update32(struct hw_endpoint * ep, uint32_t and_mask, uint32_t or_mask) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\rp2040_usb.c:111)
_hw_endpoint_buffer_control_set_value32(uint32_t value, struct hw_endpoint * ep) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\rp2040_usb.h:122)
hw_endpoint_start_next_buffer(struct hw_endpoint * ep) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\rp2040_usb.c:208)
hw_endpoint_xfer_start(struct hw_endpoint * ep, uint8_t * buffer, uint16_t total_len) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\rp2040_usb.c:240)
hw_endpoint_xfer(uint8_t ep_addr, uint8_t * buffer, uint16_t total_bytes) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\dcd_rp2040.c:185)
dcd_edpt_xfer(uint8_t rhport, uint8_t ep_addr, uint8_t * buffer, uint16_t total_bytes) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\dcd_rp2040.c:523)
usbd_edpt_xfer(uint8_t rhport, uint8_t ep_addr, uint8_t * buffer, uint16_t total_bytes) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd.c:1249)
_data_stage_xact(uint8_t rhport) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd_control.c:99)
tud_control_xfer(uint8_t rhport, const tusb_control_request_t * request, void * buffer, uint16_t len) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd_control.c:121)
process_get_descriptor(uint8_t rhport, const tusb_control_request_t * p_request) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd.c:992)
process_control_request(uint8_t rhport, const tusb_control_request_t * p_request) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd.c:715)
tud_task_ext(uint32_t timeout_ms, _Bool in_isr) (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd.c:525)
tud_task() (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\src\device\usbd.h:55)
main() (c:\Program Files\Raspberry Pi\Pico SDK v1.5.1\pico-sdk\lib\tinyusb\examples\device\uac2_headset\src\main.c:112)
```

Note that Pixel 8 never detects the headset. This problem has been driving me insane for a few months now. Finally I've seen this issue and realized the problem is in the library.. :(

HiFiPhile, 2023-11-26:
@rpascalsdl 
It's not the same issue. This one is specific to dwc2 IP, rp2040 use a different USB controller.

rpascalsdl, 2023-11-27:
@HiFiPhile thanks. I removed the comment. Will try to create an issue when I have better repro steps.