hathach/tinyusb #1982: No Mouse HID scroll events being passed

Black-FX, 2023-03-25:
### Operating System

Linux

### Board

Custom RP2040 Board

### Firmware

https://github.com/Black-FX/RP2040_Mouse

### What happened ?

I have developed a firmware for RP2040 to be used as a USB to quadrature mouse adaptor. I have the code there to handle mouse scroll events but none ever appear. 

### How to reproduce ?

1. Compile firmware and flash to an RP2040 board.
2. Connect a USB mouse using PIN8 for D+ and PIN9 for D-
3. Attempt to use scroll wheel, nothing happens
4

### Debug Log as txt file

https://gist.github.com/Black-FX/4d7c91fd831509222e58c1071626a4ee

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Ryzee119, 2023-03-25:
Mouse wheel data is not on boot protocol mode with tush uses by default (for simplicity). You need to set it to report protocol mode then parse the HID descriptor etc.
There's a function called `tuh_hid_set_protocol` to do this.

Black-FX, 2023-03-25:
Thanks, got it working :)

closed 2023-03-25 by Black-FX